RPi-Distro/repo #88: Version 9 (stretch) come with strange brcm43430 firmware

agherzan, 2017-09-11:
Downloaded a pristine raspbian image `2017-09-07-raspbian-stretch-lite.img`. 

This includes the brcm43430 firmware in the package:
```
dpkg -s firmware-brcm80211 | grep -i version
Version: 1:20161130-3+rpi2
```

The version of this firmware is `7.45.41.46`.

The package downloaded from https://packages.debian.org/stretch/firmware-brcm80211 comes with another version of the same firmware which corresponds with the old `7.45.41.26 ` version. Does anyone know how did this happen?

XECDesign, 2017-09-11:
The debian package takes the firmware that's available in linux-firmware while ours has come directly from Cypress (for the broadpwn fix, among other).

agherzan, 2017-09-11:
Got it. Where can I see the source of this version?

agherzan, 2017-09-11:
By source I mean the place were is it taken from.

XECDesign, 2017-09-11:
It was provided by Cypress directly to us. I don't know if it's publicly available anywhere else.

agherzan, 2017-09-11:
That makes sense. Could you point me to where I can download the deb modified by you with the updated version?

XECDesign, 2017-09-11:
`apt-get download firmware-brcm80211` or from here
http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_20161130-3+rpi2_all.deb

agherzan, 2017-09-11:
We would like to include it in the yocto BSP. Do you have any special licencing on this and the NVRAM configuration?

XECDesign, 2017-09-11:
I think @pelwell might know the specifics about that.

pelwell, 2017-09-11:
I am awaiting an official answer, but I'm not anticipating any objections.

agherzan, 2017-09-12:
Thanks @pelwell . I'll close this for now. Do ping me if there are any problems raised.

closed 2017-09-12 by agherzan

pelwell, 2017-09-14:
Cypress are happy for that firmware and nvram.txt to be redistributed, They are in the process of investigating a few issues for us, but I would expect that firmware or a further update to it to eventually appear in the linux-firmware repo.

agherzan, 2017-09-14:
Thanks @pelwell . Could you tell us what issues you guys reported or are aware of using this version?

pelwell, 2017-09-14:
No new issues as far as I know - just a backlog of old problems (packet loss due to ToS, AP mode issues, mailbox errors) that are receiving attention and may result in an even better firmware.

agherzan, 2017-09-14:
Thanks @pelwell once again.