RPi-Distro/repo #89: OpenGL warns "libGL error: MESA-LOADER: failed to retrieve device information" on Mesa 13.0.0 and later versions

runboy93, 2017-09-24:
I think these OpenGL warns came after Mesa got updated to 13.0.0 and this also affect later versions.

glxinfo errors

libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information

I've seen these warns on these Mesa versions:
GL_VERSION:    2.1 Mesa 13.0.0
GL_VERSION:    2.1 Mesa 13.0.6
GL_VERSION:    2.1 Mesa 17.0.0-rc2 (git-d283ec0)

I also noticed that similar problem got fixed on newer libdrm version (currently Raspbian uses 2.4.71, fixed version is 2.4.73 and later versions)
https://bugs.freedesktop.org/show_bug.cgi?id=98629#c6

XECDesign, 2017-09-25:
What problems does this cause?

runboy93, 2017-09-25:
I know it's very minor bug, but very common to see after Mesa 13.0.0 and later versions. Old libdrm version (2.4.71 and maybe older) cause this.

XECDesign, 2017-09-25:
Then the fix should eventually end up in Debian and Raspbian. We try to avoid jumping ahead when we can, so unless this causes problems, I'll have to leave it.

closed 2017-10-02 by XECDesign

spk921, 2019-04-25:
@runboy93  So how to fix this problem?

glenneroo, 2019-04-30:
Any info on how to fix this error?

Binog, 2020-01-03:
do have the same problem after updating to 2.10.0

wkoffel, 2020-05-23:
I'm also having this problem when running an Electron app on the latest version of Raspbian.  Unclear why it's marked as closed here, but for my use-case, there's an Electron app running on the main DISPLAY, but I need the command line to interact with it, so the MESA-LOADER warnings are a hassle as they clutter the shell CLI when run.  Anyone have a workaround to at least suppress these?

XECDesign, 2020-05-23:
The issue was closed because the version of libdrm we currently use is newer than 2.4.73. However, it looks like that had nothing to do with it. As far as I can tell it's only a cosmetic issue, I'll leave it as a wontfix unless somebody wants to look into it and send a patch.