RPi-Distro/repo #90: Missing kodi sources in jessie

hashashin, 2017-09-24:
I've the deb-src lines uncommented in "/etc/apt/sources.list" and "/etc/apt/sources.list.d/raspi.list", but after apt-get update I get:
```
apt-get source kodi
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to find a source package for kodi
```
Thanks.

XECDesign, 2017-09-25:
The kodi package is provided by pipplware, and I believe the .orig tarball can be found here:
http://pipplware.pplware.pt/pipplware/dists/stretch/main/binary/kodi_17.4.orig.tar.xz

I am not sure why there is no .dsc, but it's possible because it's not based on a 'proper' debian source package.

closed 2017-09-25 by XECDesign

asavah, 2017-09-27:
There is no .dsc and no .debian.tar.xz and thus no debian/rules or changelos,
it is unknown which options were used to build the package, with which custom patches (if any) and no changelog for package version bumps, eg it is unknown what changed between kodi_17.4-1~stretch and 17.4-2 , IMO this is unacceptable.

wsnipex, 2017-09-28:
IIRC they build the packages with kodi's built-in cmake deb generator instead of debuild or the like.
@nsenica can you comment?

nsenica, 2017-09-28:
Yes, it is built with cmake deb generator, the source code and build scripts are available [here](https://github.com/PIPplware/xbmc/commits/krypton_new).

This [commit](https://github.com/PIPplware/xbmc/commit/a1c21829fb68b29d6135f41d40828487fe7946bc) adds support to generate source packages to be made available on the repo.  

Last I discussed with @diogomsantos he was to make it available in raspbian repo. But I'm not sure about the current state of it.

Also, these builds are provided by pipplware and then integrated into raspbian, Kodi builds for raspberry pi afaik are not being distributed in debian nor raspbian directly.

diogomsantos, 2017-09-28:
It is pretty much as nsenica said, and I also have explained it in some forums like raspberry.org, kodi.tv and retropie.org (I am rascas in that forums by the way).

We compile and build Kodi packages now with cmake + cpack and cpack still lacks the ability to create .dsc and debian.tar.gz. 
The source code, like nsenica said is available here
https://github.com/PIPplware/xbmc/tree/krypton_new and it contains the script we use to create the packages. The diferences from the last build are the new commits.
We have the source code tarball here http://pipplware.pplware.pt/pipplware/dists/stretch/main/binary/kodi_17.4.orig.tar.xz, we said that to the Raspberry Pi Foundation guys, but they probably didn't upload it, because without the .dsc and debian.tar.gz files, you won't be able to get it through apt-get source. 

What would be really nice is if anybody with the skills could implement support for creation of .dsc, debian.tar.gz, etc stuff in cpack.

XECDesign, 2017-09-28:
The tarball isn't included in the rpi repo because we manage it using reprepro, which doesn't support tarballs which aren't referenced by an appropriate .dsc. And of course standard Debian tools wouldn't pick it up even if I dropped it in manually.