RPi-Distro/repo #92: Chromium won't start on Rpi

fusionfoto, 2017-09-30:
So this is happens both with the previous export DISPLAY=:0 advice, with a monitor and keyboard directly attached or over ssh.  I put a sym link in /usr/lib/nss to go to /usr/lib/arm*/nss too.

I'm running the latest dist-upgrade (everything is updated and upgraded). The system was working fine until these, so its clearly related to something getting upgraded.

I've uninstalled and reinstalled chromium, deleted my preferences, etc. 

pi@raspberrypi:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!                                                                        errno=2
[1:1:0929/235400.107435:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper                                                                       startup ack (0 bytes)
[2242:2272:0929/235400.332927:ERROR:nss_util.cc(712)] Error initializing NSS wit                                                                      h a persistent database (sql:/home/pi/.pki/nssdb): libsoftokn3.so: cannot open s                                                                      hared object file: No such file or directory
[2242:2272:0929/235400.333600:ERROR:nss_util.cc(201)] Error initializing NSS wit                                                                      hout a persistent database: NSS error code: -5925
[2242:2272:0929/235400.333924:FATAL:nss_util.cc(203)] nss_error=-5925, os_error=                                                                      0
#0 0x0000566a5450 <unknown>
#1 0x0000566ba708 <unknown>
#2 0x000056b1c7c4 <unknown>
#3 0x000056a65d00 <unknown>
#4 0x000056a56610 <unknown>
#5 0x0000569187a4 <unknown>
#6 0x0000568d310c <unknown>
#7 0x0000564882ac <unknown>
#8 0x000055be2d10 <unknown>
#9 0x0000566fdc24 <unknown>
#10 0x0000566f9a48 <unknown>

Any advice would be most appreciated!

thanks.

XECDesign, 2017-10-02:
Works okay here.
What's the output of `apt-cache policy chromium-browser`?

XECDesign, 2017-10-30:
No response

closed 2017-10-30 by XECDesign