RPi-Distro/repo #93: dnsmasq security vulnerabilities

OscarBarrett, 2017-10-03:
Several new vulnerabilities are detailed here: https://security.googleblog.com/2017/10/behind-masq-yet-more-dns-and-dhcp.html

The repo is currently providing `2.76-5+rpi1`.
`2.72-3+deb8u2` and `2.76-5+deb9u1` have been released for debian - could the package be updated here?

https://security-tracker.debian.org/tracker/DSA-3989-1

XECDesign, 2017-10-03:
Which repo? We provide `2.76-5+rpt1+deb9u1`

OscarBarrett, 2017-10-03:
jessie. So it's already been patched? The version shown doesn't include an update suffix (beyond `+rpi1` which it included before)

If so, feel free to close this off.

XECDesign, 2017-10-03:
It has been on Stretch, but raspberrypi.org oldstable packages don't receive updates.

XECDesign, 2017-11-01:
The package has been removes from the raspberrypi.org jessie repo a while back. I'd advise jessie users to downgrade it to the stock version by running `sudo apt-get install dnsmasq=2.72-3+deb8u`

closed 2017-11-01 by XECDesign