RPi-Distro/repo #95: Geany: 'execute' command not working - blank terminal

andrum99, 2017-10-28:
Hi folks - not sure if this is the correct place to report this issue. When using Geany on Raspbian Stretch, the 'execute' command results in a blank terminal window. Changing the 'terminal' setting in Geany (Edit -> Preferences -> Tools -> Terminal) to the following:

`terminal_cmd=x-terminal-emulator --command="/bin/sh %c"`

(per https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=191818#p1203549)

makes it work again as expected. Can this be looked at?

XECDesign, 2017-10-30:
Thanks for reporting the issue. It should be fixed now. If geany.conf has already been generated, the manual change would still be needed. If geany hasn't run before, then it should default to the correct command.

closed 2017-10-30 by XECDesign

andrum99, 2017-10-30:
Many thanks for the speedy response. I deleted ~/.config/geany and installed the updated package, which works as expected.