RPi-Distro/repo #97: Latest version of kodi core dumps on a Pi 1

andrum99, 2017-12-12:
I just applied some updates to my Pi 1 that required an reboot. Upon reboot, kodi would no longer start. The error produced is "Illegal instruction (core dumped)". I believe there has only been one update to kodi in recent months - to update to verson 17.6. I was previously using version 17.5 quite happily on this same Pi.

I have used a spare SD card to do a clean install of Raspbian, then Kodi, on this same Pi with the same result. I have also verified that I am using a version of Kodi from the archive.raspberrypi.org repository.

I have attached the crash log below. Let me know if you need any further information.

[kodi_crashlog-20171212_000624.log](https://github.com/RPi-Distro/repo/files/1549823/kodi_crashlog-20171212_000624.log)

XECDesign, 2017-12-12:
Thanks for the report.

The package is provided by Rascas of pplware and he seems to be aware of the problem:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=192499&start=50#p1244343

XECDesign, 2017-12-14:
The problem should be resolved now.

andrum99, 2017-12-14:
Yup - now working on my Pi 1. Thanks for the speedy fixa

closed 2017-12-14 by andrum99