RPi-Distro/repo #98: Raspbian Stretch: APT over SSH returns Data socket timed out

gazoo4, 2017-12-15:
The issue has been originally opened under [raspberrypi/linux](https://github.com/raspberrypi/linux/issues/2284), but it has been suggested to move it over here.

In a nutshell we are unable to access via ssh our custom package repository built by `reprepro` using stretch, while it still works with jessie. Our current investigation points to `dd` calls issued with `skip=1` argument for reading the `InRelease`, `Release` and `Packages` files which fail whereas on jessie the calls are issued with `skip=0` and there they work well.

For more details please refer to the content of the [original report](https://github.com/raspberrypi/linux/issues/2284).

bernardhalas, 2022-01-19:
I can't reproduce anymore unfortunately as I don't have the setup at hand anymore.

XECDesign, 2022-01-19:
Alright, sorry this issue was left behind for this long. I'll close it for now, but if it comes up again, please re-open and I'll put it at the top of the todo list.