RPi-Distro/repo #100: Error: Trying to overwrite '/usr/bin/scrollphathd_api', which is also in package python-scrollphathd 1.1.0

teusink, 2018-01-14:
I have an error on both a clean and freshly (but fully updated) installed Pi, and one with additional software.

Why I did try and didn't work:
- remove package from /var/cache/apt/archives before doing upgrade.
- `sudo apt-get install --reinstall python3-scrollphathd`

See below for the English error:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  python3-scrollphathd
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 18.0 kB of archives.
After this operation, 14.3 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf python3-scrollphathd all 1.1.0 [18.0 kB]
Fetched 18.0 kB in 0s (80.4 kB/s)         
Reading changelogs... Done
(Reading database ... 83564 files and directories currently installed.)
Preparing to unpack .../python3-scrollphathd_1.1.0_all.deb ...
Unpacking python3-scrollphathd (1.1.0) over (1.0.1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb (--unpack):
 trying to overwrite '/usr/bin/scrollphathd_api', which is also in package python-scrollphathd 1.1.0
Errors were encountered while processing:
 /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

The bug is also mentioned here by someone else: https://bugs.launchpad.net/raspbian/+bug/1743199

holta, 2018-01-14:
Confirmed.  This is serious as "apt update" is failing for everyone running Raspbian (With Desktop and also Raspbian Lite?)

Many more bug reports here:
https://bugs.launchpad.net/raspbian/+bug/1743199

Wikibear, 2018-01-14:
I can confirm this bug. I get this error also.  I remove package and upgrade again and was fine and this package will be installed automatically without my doing after upgrade again.

FatalXCept10n, 2018-01-14:
I got the error as well. This was on a fresh install of the latest Stretch from a noobs-lite. I ran:
sudo apt-get update
Which completed with no errors. 
I then ran:
sudo apt-get upgrade
And got the error. I tried removing the archive from var/cache which was a deb file, and I deleted the python-3- scrollphathd-api file and still could not install scrollphathd. Hardware: RPi2B with GPS hat from adafruit and raspberry pi official 7” touchscreen. Reputable and reliable fast storage micro sd card. 
I have not been able to reinstall the package and it has not tried to install it during subsequent apt-get upgrades.

XECDesign, 2018-01-14:
ping @Gadgetoid

Wikibear, 2018-01-14:
Try:
sudo apt remove -y python3-scrollphathd
sudo dpkg --configure -a
sudo apt install python3-scrollphathd -y

Do the trick for me.

ilkevinli, 2018-01-14:
Still didn't work for me:

pi@raspberrypi:~ $ sudo dpkg --configure -a
pi@raspberrypi:~ $ sudo apt install python3-scrollphathd -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Suggested packages:
  python3-psutil
The following NEW packages will be installed:
  python3-scrollphathd
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/18.0 kB of archives.
After this operation, 247 kB of additional disk space will be used.
(Reading database ... 166593 files and directories currently installed.)
Preparing to unpack .../python3-scrollphathd_1.1.0_all.deb ...
Unpacking python3-scrollphathd (1.1.0) ...
dpkg: error processing archive /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb (--unpack):
 trying to overwrite '/usr/bin/scrollphathd_api', which is also in package python-scrollphathd 1.1.0
Errors were encountered while processing:
 /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@raspberrypi:~ $

filthycoding, 2018-01-14:
@Wikibear Didn't help for me in any way, with none of the two packages.

aletron, 2018-01-14:
@Wikibear same, the method did not work.

davidhowell-tx, 2018-01-14:
I'm having the same issue.  I tried removing both python and python3 versions of scrollphathd and installing python, then python3 version.  Same error occurs:

- sudo apt-get remove python3-scrollphathd
- sudo apt-get remove python-scrollphathd
- sudo rm /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb
- sudo rm /var/cache/apt/archives/python-scrollphathd_1.1.0_all.deb
- sudo apt-get install python3-scrollphathd
- sudo apt-get install python-scrollphathd
- dpkg: error processing archive /var/cache/apt/archives/python-scrollphathd_1.1.0_all.deb (--unpack):
 trying to overwrite '/usr/bin/scrollphathd_api', which is also in package python3-scrollphathd 1.1.0
Errors were encountered while processing:
 /var/cache/apt/archives/python-scrollphathd_1.1.0_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

davidhowell-tx, 2018-01-14:
I ran the following to force dpkg to overwrite the noted scrollphathd_api file, but unsure if this causes issues with either of the programs:

sudo dpkg -i --force-overwrite /var/cache/apt/archives/python-scrollphathd_1.1.0_all.deb
(Reading database ... 123211 files and directories currently installed.)
Preparing to unpack .../python-scrollphathd_1.1.0_all.deb ...
Unpacking python-scrollphathd (1.1.0) ...
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: trying to overwrite '/usr/bin/scrollphathd_api', which is also in package python3-scrollphathd 1.1.0
Setting up python-scrollphathd (1.1.0) ...

reinvl, 2018-01-15:
I am getting the same error. Looking for a solution.

0n3-70uch, 2018-01-15:
Hello,
@Wikibear Didn't work for me too.

0n3-70uch, 2018-01-15:
Solution found here:
https://www.raspberrypi.org/forums/viewtopic.php?t=202523

Just type the following:
`cd /var/cache/apt/archives 
sudo dpkg -i --force-overwrite python3-scrollphathd_1.1.0_all.deb`

teusink, 2018-01-15:
@0n3-70uch That is a workaround imo :). A solution is that the error won't come back after reinstalling the packages or when doing the update :).

0n3-70uch, 2018-01-15:
Yeah you are right... :-)

nathan-weinberg, 2018-01-15:
@0n3-70uch Workaround fixed the issue for me as well -  thanks!

JorinL, 2018-01-15:
@0n3-70uch yep that's it!

XECDesign, 2018-01-16:
A new version of the package has been uploaded today. Does it resolve the issue?

filthycoding, 2018-01-16:
@XECDesign No, it doesn't. Both packages still conflict with each other in the exact same way.

turboc1208, 2018-01-16:
it looks like scrollphathd is for some type of external scrolling display.  Since I don't have that device I just uninstalled the python3-scrollphathd and problem solved.  I'm not sure why it was being installed in the first place.

XECDesign, 2018-01-16:
Thanks for checking. I've forwarded the issue on more directly.

Gadgetoid, 2018-01-16:
@filthycoding @XECDesign I don't understand why python3-scrollphathd_1.1.1 and python-scrollphathd_1.1.1 have not fixed this issue. I removed the binary endpoint and republished as soon as I saw this on the forums today.

Installing these two packages locally and inspecting their contents with `dpkg -L` does not list the path `/usr/bin/scrollphathd_api`

You can obtain the 1.1.1 deb files here:  https://github.com/pimoroni/scroll-phat-hd/releases/tag/v1.1.1

filthycoding, 2018-01-17:
Okay, for some reason, the issue is fixed now, but wasn't before (even though both versions were updated to 1.1.1 and I tried installing one after another, just now it suddenly worked on its own).

lukasl1991, 2018-01-17:
I had exactly the same problem. Do I understand correctly that I could now try it again without doing
`cd /var/cache/apt/archives
sudo dpkg -i --force-overwrite python3-scrollphathd_1.1.0_all.deb`

Gadgetoid, 2018-01-17:
@lukasl1991 

Yes, version 1.1.1 should have made its way through the system now, try:

```
sudo apt update
sudo apt upgrade
```

It should pull the new 1.1.1 version instead of 1.1.0. Bearing in mind 1.1.0 might otherwise be lodged in your cache.

lukasl1991, 2018-01-17:
@Gadgetoid

Yes, it works now! Thank you ;-)

0n3-70uch, 2018-01-17:
@Gadgetoid Hi I received the same error like before. But this is running:
```
sudo apt update
sudo apt dist-upgrade
```
It's solved now... thanks.

XECDesign, 2018-01-17:
Thanks everyone

closed 2018-01-17 by XECDesign

RobertOnGit, 2018-01-19:
👍