RPi-Distro/repo #103: std::atomic has race condition with g++/gcc on a raspberry pi 3B+

gpmfuchs, 2018-04-11:
std::atomic is not working with g++/gcc in raspberry pi 3B+ running raspbian.

it seems to behave like a normal variable with race conditions.

I've used "g++ (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516"
(I have not tested yet if the problem comes from upstream)

atomics are working with clang++/llvm on the same system (clang++ pierror.cpp -std=c++11 -l pthread -o piok).
atomics are working with raspberry pi 2B running arch
atomics are working on x64

[atomic.zip](https://github.com/RPi-Distro/repo/files/1897636/atomic.zip)

(this bug was missreported to the kernel https://github.com/raspberrypi/linux/issues/2504)

XECDesign, 2018-04-11:
Ideally, the issue should be reported here https://bugs.launchpad.net/raspbian, since we don't provide a custom version of gcc in archive.raspberrypi.org.

If you can confirm that the problem comes from the upstream debian package, then that would push the problem further back to https://www.debian.org/Bugs/Reporting

This is well above anything I have the experience to deal with quickly.

popcornmix, 2018-04-11:
Cross compiling the example code with https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf does pass the test. I see the failure with g++ from raspbian stretch.

So it does feel like a compiler bug, rather than broken locking support in hardware/kernel.

XECDesign, 2018-08-15:
Closing due to lack of activity.

closed 2018-08-15 by XECDesign

rsalinas, 2018-11-19:
This problem is still happening. 

You can make it manifest with this code: 

```
#include <atomic>
#include <thread>
#include <cstdio>

int main(int argc, char **argv) {
    constexpr int itemsPerThread = 10000;
    constexpr int threadCount = 20;
    constexpr int idCount = itemsPerThread * threadCount;
    typedef uint64_t testedType;

    auto ids = new testedType[threadCount][itemsPerThread];
    std::thread threads[threadCount];
    std::atomic<testedType> atomicId{0};

    for (int th = 0; th < threadCount; th++) {
        threads[th] = std::thread([&, th]() {
            for (size_t i = 0; i < itemsPerThread; ++i) {
                ids[th][i] = atomicId++;
            }
        });
    }
    auto seenIds = new bool[threadCount*itemsPerThread]{};
    for (int i = 0; i < threadCount; i++) {
        threads[i].join();
        for (int j = 0; j < itemsPerThread; j++) {
            testedType id = ids[i][j];
            if (seenIds[id]) {
                fprintf(stderr, "Error: Your std::atomic is broken: %lu :(\n", id);
                exit(1);  // Forget the running threads
            }
            seenIds[id] = true;
        }
    }
    delete[] ids;
    fprintf(stderr, "Success :)\n");
}
```