RPi-Distro/repo #106: Limburgish as a language option in Belgium?

mraeymaekers, 2018-07-30:
After installing Raspbian 2018-06-27, I am presented with language options.
I choose Belgium as the country and 4 language options are shown, included Limburgish.
I see same when going to Raspberry pi Configuration and clicking "set Locale...", there is a Locale for Limburgish (BE li)
I like a good joke like everyone else, but please note that Limburgish is not a belgian language. It is a flemish dialect spoken by people in the province of Limburg. Just like each of the 5 flemish provinces people have their own accents.

XECDesign, 2018-07-30:
@spl237 maybe 'locale' would be more appropriate than 'language'?

spl237, 2018-07-30:
Technically, in Linux, a locale is a combination of a country and a language, so the correct label for that box is "language". 

If the details of this particular locale are incorrect, it is incorrect on Debian Linux, not just on us, and so needs to be fixed in Debian. Once it is fixed in there, the fix will reach Raspbian.

All I am doing is taking verbatim the contents of the existing Linux locale databases and parsing them into our GUI - I'm not making any political, social or economic judgements thereon... ;)

closed 2018-07-30 by spl237

XECDesign, 2018-07-30:
The format is "[language[_territory][.codeset][@modifier]]". Language and territory come from 2 alpha codes from ISO 639 and ISO 3166, respectively.

li_BE is valid in terms of both standards.

XECDesign, 2018-07-30:
Oh yeah, that wasn't meant to be a pedantic correction. Was just adding what I found in my extensive skimming through the wikipedia page.