RPi-Distro/repo #107: dhcpcd.service is not restarted after SIGSEGV

brubbel, 2018-08-06:
Background:
(Slightly related to this [issue](https://github.com/raspberrypi/linux/issues/2453), where the brcmf driver crashes every few hours/days.)
1. Interface wlan0/uap0 (!= eth0) goes down, carrier lost, whatever.
2. dhcpcd.service enters failed state.
3. eth0 network goes down (no gateway, dns).

Expected behaviour:
Systemd restarts dhcpcd.service

Cause:
`Restart=always (?)`
is missing in /lib/systemd/system/dhcpcd.service

While dhcpcd should not segfault in the first place when brcmfmac crashes, _is there a good reason NOT to restart when dhcpcd goes down_?

```
● dhcpcd.service - dhcpcd on all interfaces
   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dhcpcd.service.d
           └─wait.conf
   Active: failed (Result: signal) since Thu 2018-08-02 21:17:46 UTC; 3 days ago
 Main PID: 486 (code=killed, signal=SEGV)
      CPU: 54.627s

Aug 02 21:17:44 raspberrypi dhcpcd[486]: wlan0: deleting route to 192.168.0.0/24
Aug 02 21:17:45 raspberrypi dhcpcd[486]: uap0: adding route to 96.112.72.0/21
Aug 02 21:17:45 raspberrypi dhcpcd[486]: if_route (ADD): Invalid argument
Aug 02 21:17:46 raspberrypi dhcpcd[486]: uap0: carrier lost
Aug 02 21:17:46 raspberrypi systemd[1]: dhcpcd.service: Main process exited, code=killed, status=11/SEGV
Aug 02 21:17:46 raspberrypi dhcpcd[14390]: dhcpcd not running
Aug 02 21:17:46 raspberrypi dhcpcd[14390]: sending signal TERM to pid 486
Aug 02 21:17:46 raspberrypi systemd[1]: dhcpcd.service: Control process exited, code=exited status=1
Aug 02 21:17:46 raspberrypi systemd[1]: dhcpcd.service: Unit entered failed state.
Aug 02 21:17:46 raspberrypi systemd[1]: dhcpcd.service: Failed with result 'signal'.
```

XECDesign, 2018-08-06:
Perhaps it should restart, but something else is (also) not right here.

dhcpcd doesn't die if the interface goes down. If it's segfaulting then there's some other issue which is worth investigating.

XECDesign, 2018-08-06:
Since dhcpcd doesn't stop if an interface goes down, and the linked issue seems to (only?) happen with nexmon patched firmware, I'd be a little cautious before rushing to change things.

What would restarting dhcpcd accomplish in the above scenario? Wouldn't it crash over and over until systemd stops trying?

One potential issue is that it may be masking a more serious problem, so if a beginner reports that their network is unstable, it would take a while to realise that dhcpcd is crashing every minute, for example.

brubbel, 2018-08-06:
This is happening on unpatched firmware, and indeed there are serious issues with the brcmf firmware which I don't see solved in the near future.
Maybe the best solution would be that I limit the dhcpcd to listen only on eth0, instead of grasping every interface it sees. (uap0, docker0/1, etc...)

XECDesign, 2018-08-06:
> Edit: systemd allows to set a limit on the number of unsuccessful restarts if I remember correct.

Yes, it has start rate limiting with sensible defaults, but that won't help if it crashes randomly.

I had a look at what other distros do and it their service file are pretty close to ours, without a Restart= line.

I'm leaning towards "if you're having issues with the wireless interface, disable it until it's fixed or add your own workaround". I don't like the idea of restarting the service like there's no problem and potentially causing other issues which would be harder to trace back to the root cause.

I'll ponder over it and leave it open for now in case somebody else at Raspberry Pi wants to chime in, in case I'm wrong.

brubbel, 2018-08-06:
There is no rush here, I was just trying to find out why some (remote) systems randomly disconnect. At least I'm closing in on the cause now.
For me, running a separate dhcpcd process for eth0 and wireless will the best way to avoid this chain reaction of one fault leading to the next.

XECDesign, 2018-08-23:
I think I'll close this one based on lack of more input and that other distros do it the same way.

closed 2018-08-23 by XECDesign

jmaurice, 2019-07-19:
Perhaps adding a delay, I would request that `Restart=always` be added, dhcpcd is a critical service and for some reason if it crashes occasionally I would like it to restart so I can still remotely access my RPI.

jkuek, 2020-05-26:
I just encountered a similar issue on unpatched firmware. I'm running the latest patched Raspbian Stretch.

dhcpcd.service exited with a SEGV when I power-cycled the switch the RPI was connected to.

I had to manually restart the service to resume network.

fracture-point, 2020-10-28:
I also encountered this issue and the only resolution was to reboot.  I'm running 18 containers on a pi4, latest Raspbian, stock firmware.  First I got a few of these errors:
```Oct 26 23:33:30 raspberrypi systemd-udevd[1681]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Oct 26 23:33:30 raspberrypi systemd-udevd[1681]: Could not generate persistent MAC address for vethc0ea94b: No such file or directory
Oct 26 23:33:30 raspberrypi containerd[584]: time="2020-10-26T23:33:30.189105685-04:00" level=info msg="shim containerd-shim started" address=/containerd-shim/2eea9e572e2b5a07ad975f50a987d9dd7c47efde325ce8fe33d1e67b88711fff.sock debug=false pid=1889
```

Followed by DHCP giving up....
`Oct 26 23:33:30 raspberrypi systemd[1]: dhcpcd.service: Main process exited, code=killed, status=11/SEGV`

Anddddd finally the complete implosion of network connectivity:
```
Oct 26 23:33:31 raspberrypi avahi-daemon[388]: Interface wlan0.IPv4 no longer relevant for mDNS.
Oct 26 23:33:31 raspberrypi avahi-daemon[388]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 10.0.1.179.
Oct 26 23:33:31 raspberrypi avahi-daemon[388]: Withdrawing address record for x.x.x.x on wlan0.
Oct 26 23:33:31 raspberrypi systemd[1]: dhcpcd.service: Failed with result 'signal'.
```

Does anyone have a workaround / interim solution?  I have been playing whack-a-mole dodging critical connectivity issues on both eth and wlan in RPi.

yosemitebandit, 2021-02-09:
Not sure if this is possible for you, but I added `denyinterfaces veth*` to `/etc/dhcpcd.conf` and that helped.  Further notes [here](https://github.com/raspberrypi/linux/issues/4092/).