RPi-Distro/repo #109: libwebsockets8 is linked against libssl1.0.2

wjoost, 2018-08-18:
In stretch, package libwebsockets8 is linked against libssl 1.02. However, the corresponding development package libwebsockets-dev requires installation of libssl-dev in version 1.1.

pi@raspberrypi:~ $ apt-cache depends libwebsockets8 libwebsockets-dev
libwebsockets8
  Depends: libc6
  Depends: libev4
  Depends: libssl1.0.2
  Depends: libuv1
  Depends: zlib1g
libwebsockets-dev
  Depends: libwebsockets8
  Depends: libev-dev
  Depends: libssl-dev
  Depends: libuv1-dev
  Depends: zlib1g-dev

This causes problems with programs which want to use both libwebsockets8 and libssl. I think this might affect package mosquitto:

pi@raspberrypi:~ $ apt-cache depends depends mosquitto libwebsockets8
mosquitto
  Depends: adduser
  Depends: libuuid1
  Depends: lsb-base
  Depends: libc6
  Depends: libssl1.1
  Depends: libwebsockets8
  Depends: libwrap0
  Suggests: apparmor
libwebsockets8
  Depends: libc6
  Depends: libev4
  Depends: libssl1.0.2
  Depends: libuv1
  Depends: zlib1g

On x86_64, libwebsockets8 is linked against libssl v.1.:

root@vtest4:~# apt-cache depends libwebsockets8 libwebsockets-dev
libwebsockets8
  Depends: libc6
  Depends: libev4
  Depends: libssl1.1
  Depends: libuv1
  Depends: zlib1g
libwebsockets-dev
  Depends: libwebsockets8
  Depends: libev-dev
  Depends: libssl-dev
  Depends: libuv1-dev
  Depends: zlib1g-dev

I think, libwebsockets8 should be build and linked against libssl v1.1 on raspbian, too.

XECDesign, 2018-08-20:
Technically, this is a raspbian.org issue, which should be filed here https://bugs.launchpad.net/raspbian

However, since the fix seems to be straight forward rebuild, it's not much of a problem.

XECDesign, 2018-08-20:
The fix should appear in apt shortly. Please test and close the issue if it's resolved.

wjoost, 2018-08-21:
Currently, in apt-cache stills shows a version with the described problem.

XECDesign, 2018-08-22:
Looks okay from here, but there are issues with the mirrors (https://github.com/RPi-Distro/repo/issues/111).

It took me a couple of refreshes to hit the right mirror to see the new files.

What happens when you run this?
```
sudo apt update
apt policy libwebsockets8
```

wjoost, 2018-08-22:
Looks good now:

apt policy libwebsockets8
libwebsockets8:
  Installed: 2.0.3-2+b1~rpt1
  Candidate: 2.0.3-2+b1~rpt1
  Version table:
 *** 2.0.3-2+b1~rpt1 500
        500 http://archive.raspberrypi.org/debian stretch/main armhf Packages
        100 /var/lib/dpkg/status
     2.0.3-2 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages

ldd /usr/lib/arm-linux-gnueabihf/libwebsockets.so.8 | grep ssl
	libssl.so.1.1 => /usr/lib/arm-linux-gnueabihf/libssl.so.1.1 (0x76eb6000)

closed 2018-08-22 by wjoost