RPi-Distro/repo #110: linux-perf for the 4.14 kernel

pcollinson, 2018-08-20:
Can this be added to the stretch release please.

XECDesign, 2018-08-30:
Is this something that needs to be rebuilt for every minor version of the kernel or will a single package work for all 4.14 kernels?

pcollinson, 2018-08-30:
Sorry - I am not really sure. I was wanting to use perf to track down packet loss and was pointed in the general direction of perf. I haven't hacked on Unix kernel code since about 1977.

However, current versions of perf are distributed as /usr/bin/perf_4.9 - and the actual perf command is a shell script which checks machine version against available binaries. So my guess is that a version called perf_4.14 would do the trick.

XECDesign, 2018-08-30:
Alright, I'll add it to the list, but it fairly low priority.

superfrink, 2020-07-08:
@XECDesign , is there a way for someone to contribute a patch to update the build version for linux-perf?  Thanks.

XECDesign, 2020-07-08:
In debian, linux-perf comes from the linux source package. We don't use that yet, so there's not much we can do for now. It has been on my todo list forever, but something more important always comes along.

If somebody wants to help, I think the debian kernel package needs to be updated to generate each of the pi kernels without unnecessary debian patches to minimise the delta from the kernels distributed by rpi-update, a common package for the overlays and so on and the legacy raspberrypi-kernel package should pull in all the new kernels. It would also need to handle installs to /boot by copying the files from another location in postinst and make sure all the initramfs stuff is handled properly.