RPi-Distro/repo #112: dhcpcd5 no longer built with libudev-dev

burtyb, 2018-09-09:
Since upgrading from dhcpcd5_6.11.5-1+rpt5 to dhcpcd5_6.11.5-1+rpt6 I have intermittent issues renaming Ethernet interfaces.

Sep  9 15:19:33 cnat systemd-udevd[6626]: could not rename interface '38' from 'eth2' to 'ethpi1': Device or resource busy
Sep  9 16:38:24 cnat systemd-udevd[4006]: could not rename interface '18' from 'eth1' to 'ethpi3': Device or resource busy
Sep  9 16:38:26 cnat systemd-udevd[4161]: could not rename interface '20' from 'eth2' to 'ethpi1': Device or resource busy
etc.

It looks like dhcpcd5_6.11.5-1+rpt6 hasn't been compiled with the build dependency libudev-dev as it's missing the "./lib/arm-linux-gnueabihf/dhcpcd/dev/udev.so" file from the package (which exists when configure finds libudev).

Without udev support the old race issue when renaming Ethernet devices has returned.

Manually recompiling dhcpcd5_6.11.5-1+rpt6 with libudev-dev correctly allows the renaming of Ethernet devices by udev rules.

Can this change please be reverted to allow udev to rename Ethernet devices when using dhcpcd5?

XECDesign, 2018-09-09:
Well spotted, thanks. Should be sorted tomorrow.

XECDesign, 2018-09-10:
Fixed dhcpcd package should show up in the repo shortly, if it hasn't already.

burtyb, 2018-09-10:
All looks good with the new package, thanks :).

closed 2018-09-12 by XECDesign

burtyb, 2021-11-17:
It looks like this issues has returned for some architectures.

All amd64/arm64/i386/armhf versions of dhcpcd5_8.1.2-1+rpt1 were built with udev but for dhcpcd5_8.1.2-1+rpt2 only dhcpcd5_8.1.2-1+rpt2_armhf.deb is built with udev, amd64/arm64/i386 don't have it.

Can it be added back for at least arm64 as I'm seeing problems again when trying to rename a network interface in bullseye.