RPi-Distro/repo #116: libcec-dev not installable.

plugwash, 2018-11-12:
Just had https://bugs.launchpad.net/raspbian/+bug/1802820 reported.

The cause seems to be that you ship a newer version of libcec than we do, but you use a different dev package name. 

Please adjust your packaging to build a libcec-dev package in addition to your libcec4-dev package. I suggest building the libcec-dev package as a dummy package depending on libcec4-dev.

diogomsantos, 2018-11-13:
@XECDesign Sorry for the late response. I think building a libcec-dev dummy package is the best solution, because libcec4-dev is relatively new package( about 2 years ago), added by request from a Team Kodi Ubuntu PPA dev, not sure why. 
Can you create a libcec-dev or you prefer another solution? I can do it if you want.

XECDesign, 2018-11-15:
Going to be a bit busy for a few days, but I'll add something next week then.

XECDesign, 2018-11-16:
That libcec package is a mess so hats off to you for getting it to build in the first place. I've taken packages from deb-multimedia, and put them in the 'untested' component for now.

diogomsantos, 2019-03-25:
Sorry for the late response, but libcec-dev was updated some time ago, this can most likely be closed now.

XECDesign, 2019-03-26:
Thanks, Diogo.

closed 2019-03-26 by XECDesign