RPi-Distro/repo #118: Mathematica/TightVNC Interaction, seeming 'Raspbian-Stretch' Version Dependent

mrwgx3, 2018-12-12:
[linux_post]: https://github.com/raspberrypi/linux/issues/2783 "/raspberrypi/linux/issues/2783"
This issue [orginally posted here][linux_post], and moved at suggestion of @popcornmix
### Basic Infos

#### Hardware

```
Hardware:     Raspberry Pi Model 3B v1.2
Computer:     Dell, Intel Core i7 CPU, 870 @ 2.93Ghz,  16.0 GB RAM
```
#### OS, Software, Windows Side
``` 
OS:          Windows 10 Professional
Viewer:      TightVNC Viewer for Windows, version 2.8.11
             (fresh install)
RemoteHost:  192.168.1.131::5901, IP statically assigned by router
```
### Description

Last year, my friend gave my a RPI3 for Xmas, and set it up to operate headless so I could use Mathematica on my network. Everything worked fine until we updated my RPI3; after the 'sudo update/upgrade/reboot' cycle, Mathematica now terminates after running a notebook or mousing over certain areas of the Wolfram Mathematica 'welcome' screen.

### Further Investigation

To help isolate and reproduce this problem, separate 8GB SD cards were prepared with the following 'raspian-stretch' images:
```
2018-11-13-raspbian-stretch-full.img
2017-11-29-raspbian-stretch.img
```

Running the RPI3 with monitor/mouse/keyboard (headed), the keyboard type and time-zone offset were set. A terminal window was next opened and 'TightVNC Server' was installed on each system:

```sudo apt-get install tightvncserver```

After the install, 'tightvncserver' was ran to set the viewer password. Following a reboot, a terminal window was opened again to start 'tightvncserver' in preparation for 'headless' operation:

```tightvncserver :1```

### Observations

Running Mathematica for both systems in 'headed' mode had no problems (test notebooks were supplied by thumb-drive).

1. The '2017-11-29' image had no issues running Mathematica in 'headless' mode.

2. The '2018-11-13' image had the same problems encountered earlier after upgrading the original RPI3 install. As the mouse cursor position seemed to affect how soon Mathematica terminated, the cursor was move to the upper-right desktop edge after each restart of Mathematica. Both the 'notebook' and 'welcome' window screens were then explored after the kernel finished loading:

   * Mousing over the notebook body would not terminate Mathematica, but a single click did so.

   * A notebook could be run if one carefully moused into the menu area and selected options while keeping the mouse cursor within the pull-down menus. After the notebook executed, mousing into the notebook area would end the session.

   * Mousing into the bottom-half of the 'welcome screen' and even clicking the selection buttons created no problems. Mousing into either the title area or red separator below the blue menu bar, however, would immediately terminate Mathematica.

### Commentary

From my limited GUI programming experience, the problem has the 'feel' of a corrupted or disconnected event-handler associated with mouse-cursor context or 'focus'.

The Mathematica versions encountered were:
```
11/13/2018 build
TightVNC Server Version:  1.3.9
Mathematica Version:     11.3.0 (May 23, 2018)

11/29/2017 Build
TightVNC Server Version:  1.3.9
Mathematica Version:     11.0.1.0 (Jan 17, 2017)
```
If we assume these versions are close enough as not to be an issue, it would seem the problem lies (at a minimum) within the '11/13/2018' distribution, as the 'TightVNC' server versions are the same.

Good luck to you in resolving this issue.

P.S. The following tests were also ran with the extant 'RealVNC' server de-installed before the installation of 'TightVNC'; the test results were identical.

XECDesign, 2019-09-17:
Is this still a problem in the latest buster image?

mrwgx3, 2019-09-17:
Hello!
Surprised to see a response!
I haven't upgraded my Pi3  since this problem arose, still running the 11/19/2017 build. Has a fix been found, and can/should I make an attempt to upgrade the distro?

XECDesign, 2019-09-17:
No, but there have been a few Mathematica releases since then and it may be something they have already fixed. If it can bee confirmed to still be an issue on the latest image, I can forward it on.

closed 2019-12-13 by XECDesign