RPi-Distro/repo #120: Raspbian Stretch: package php-file has unmet dependency

cernekj, 2019-01-18:
```console
user@host:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

user@host:~$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

user@host:~$ sudo apt install php-file
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 php-file : Depends: php5-common (>= 4.3.0) but it is not installable
E: Unable to correct problems, you have held broken packages.

user@host:~$ cat /etc/apt/sources.list
deb http://ftp.sh.cvut.cz/raspbian/ stretch main contrib non-free rpi
```

cernekj, 2019-01-18:
First "apt upgrade" should show "apt update". My fault, copied wrongly, my apologies.

```console
user@host:~ $ sudo apt update
Hit:1 http://ftp.sh.cvut.cz/raspbian stretch InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
```

XECDesign, 2019-01-21:
This issue tracker is for packages from the 'foundation' repo (archive.raspberrypi.org), not Raspbian itself. It looks like the version of php-file is an older version for php 5 rather than php 7.

Their bug tracker is here: https://bugs.launchpad.net/raspbian/

closed 2019-01-21 by XECDesign