RPi-Distro/repo #121: wpa_supplicant flooding syslog with "Failed to initiate sched scan"

marcelser, 2019-01-30:
Hi,

I'm using a Raspberry Pi Zero W with an attached USB Ethernet Adapter. After a clean installation of Raspbian Strecht and setting up eth0 + wlan0 everything seemed fine. I then also did setup bonding on eth0 + wlan0 to make use of whatever is available where eth0 is preferred interface.  By accident I was looking in the syslog and seeing this messages. As you can see they are appearing approx every 6 seconds and were filling up my log. I might add that atm the pi is in a location where theres no AP to connect to. Nevertheless this looks like a more serious issue with driver or so. To make those messages go away I had to temporarily tear down wlan0 with "ifdown wlan0" and then it stopped.

Here's a short snippet of the log output:
```
Jan 30 07:29:02 pi-tempalarm wpa_supplicant[401]: wlan0: Failed to initiate sched scan
Jan 30 07:29:08 pi-tempalarm wpa_supplicant[401]: wlan0: Failed to initiate sched scan
Jan 30 07:29:14 pi-tempalarm wpa_supplicant[401]: wlan0: Failed to initiate sched scan
Jan 30 07:29:19 pi-tempalarm wpa_supplicant[401]: wlan0: Failed to initiate sched scan
Jan 30 07:29:25 pi-tempalarm wpa_supplicant[401]: wlan0: Failed to initiate sched scan
Jan 30 07:29:30 pi-tempalarm wpa_supplicant[401]: wlan0: Failed to initiate sched scan
Jan 30 07:29:36 pi-tempalarm wpa_supplicant[401]: wlan0: Failed to initiate sched scan
```

and here's ma /etc/network/interfaces:
`# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

auto lo
iface lo inet loopback

auto eth0
allow-hotplug eth0
iface eth0 inet manual
        bond-master     bond0
        bond-mode       active-backup

auto wlan0
allow-hotplug wlan0
iface wlan0 inet manual
        wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
        bond-master     bond0
        pre-up sleep    4
        bond-mode       active-backup

auto bond0
iface bond0 inet dhcp
    #hwaddress             b8:27:eb:79:39:6b
    bond-slaves           none
    bond-primary          eth0
    bond-mode             active-backup
    bond-miimon           200
    bond-fail_over_mac    none
    bond-primary_reselect always
    bond-updelay          200
    bond-downdelay        0

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d
`

and the wpa_supplicant (of course with replaced ssid + psk):
`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=CH

network={
     ssid="yyyyy"
     psk="xxxxxxx"
     key_mgmt=WPA-PSK
}`

what's wrong with wpa_supplicant here. Is it normal that it starts spitting out errors if not connected to wlan0?

ezequielarevalo-natgeo, 2019-07-31:
I am facing the same issue with raspbian lite on raspberry zero w

xxauroraxx, 2019-12-07:
```
pi@WATERSERVER:~ $ cat /etc/network/interfaces
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

pi@WATERSERVER:~ $ ls /etc/network/interfaces.d

pi@WATERSERVER:~ $ sudo cat /etc/wpa_supplicant/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="REMOVED"
    psk="REMOVED"
    key_mgmt=WPA-PSK
}

pi@WATERSERVER:~ $ cat /proc/device-tree/model
Raspberry Pi 3 Model B Rev 1.2

pi@WATERSERVER:~ $ sudo wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf
Successfully initialized wpa_supplicant
wlan0: Failed to initiate sched scan
wlan0: Failed to initiate sched scan
wlan0: Failed to initiate sched scan
^Cnl80211: Failed to open /proc/sys/net/ipv4/conf/p2p-dev-wlan0/drop_unicast_in_l2_multicast: No such file or directory
nl80211: Failed to set IPv4 unicast in multicast filter
nl80211: Failed to open /proc/sys/net/ipv4/conf/p2p-dev-wlan0/drop_unicast_in_l2_multicast: No such file or directory
nl80211: Failed to set IPv4 unicast in multicast filter
nl80211: deinit ifname=p2p-dev-wlan0 disabled_11b_rates=0
p2p-dev-wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
wlan0: CTRL-EVENT-TERMINATING
```

XECDesign, 2019-12-09:
@pelwell Any idea?

pelwell, 2019-12-09:
Are there any corresponding messages in the kernel (`dmesg`) log?

brubbel, 2020-03-16:
Additional info: happens when **not associated** and wpa_supplicant contains a network that is **out of range**.

claymore666, 2020-12-03:
I had the same issue on a RPi4. Worked just yesterday. Further analysis revealed also a CTRL-EVENT-ASSOC-REJECT with Status 16 in WPA Supplicant, as described [here](https://www.raspberrypi.org/forums/viewtopic.php?t=274715). After unplugging the HDMI connector the connection to the AP was directly established successfully. I am using console only, no graphics.

ejohnson54, 2021-02-14:
I grappled with this problem for quite a while on a pi 0 with wifi.  It has been a royal pain!  I found no good solutions on the web.  My problem was ultimately solved by using a generic /etc/wpa_supplicant/wpa_supplicant.conf as follows:

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="REMOVED"
    psk="REMOVED"
    key_mgmt=WPA-PSK
}

then adding these two lines to my /etc/rc.local before anything else:

service wpa_supplicant stop
wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf

Not very elegant but it works consistently for me.  Use sudo to edit everything :)

XECDesign, 2022-01-19:
Is this still an issue in Bullseye?

prk0ghy, 2022-03-10:
I have a similar issue. my device config is
```
root@raspberrypi:/home/pi# cat /etc/network/interfaces.d/wlan0
auto wlan0
  iface wlan0 inet dhcp
    wireless-essid ********
    allow-hotplug wlan0

root@raspberrypi:/home/pi# cat /etc/wpa_supplicant/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
p2p_disabled=1
country=CN
network={
        ssid="**************"
        psk="**************"
}
´´´
The raspi is in china.

darth-veitcher, 2023-09-09:
My issue stemmed from a typo in the SSID - they're case sensitive and so not only is the network not found but also the calculated `wpa_passphrase` changes in your config.

Eg. case in point below

```zsh
me@pi:~ $ wpa_passphrase mySSID supersecret

network={
	ssid="mySSID"
	#psk="supersecret"
	psk=9b1afedddc66560fe18d5d55ae7574cc80f05f83edc77db109b3bb62d1548baf
}
```

```zsh
me@pi:~ $ wpa_passphrase myssid supersecret
network={
	ssid="myssid"
	#psk="supersecret"
	psk=6f68e8cb0be6a5877fa265aa80402d6a96b77612ec6e15da97f0da40f0248c24
}
```

I didn't need to do anything special other than update the `said` and `psk` but that really threw me for a while (and the error message is unhelpfully opaque).

gearhead, 2024-01-01:
Getting the same result with bookworm. I am using connman and wpa_supplicant and get tons of this:
```
Jan 01 10:55:58 alarm wpa_supplicant[7399]: wlan0: Failed to initiate sched scan
Jan 01 10:56:39 alarm wpa_supplicant[7399]: wlan0: Failed to initiate sched scan
Jan 01 10:56:43 alarm wpa_supplicant[7399]: wlan0: Failed to initiate sched scan
Jan 01 10:56:51 alarm wpa_supplicant[7399]: wlan0: Failed to initiate sched scan
Jan 01 10:57:00 alarm wpa_supplicant[7399]: wlan0: Failed to initiate sched scan
Jan 01 10:57:27 alarm wpa_supplicant[7399]: wlan0: Failed to initiate sched scan
```
contents of my wpa_supplicant.conf
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US
```
when I issue 'connmanctl scan wifi', it scans and returns ssids and I can connect.

How to get rid of this warning?

ghollingworth, 2024-01-02:
This looks like a configuration issue:

https://askubuntu.com/questions/821406/wpa-supplicant949-wlan0-failed-to-initiate-sched-scan#:~:text=Typically%20this%20problem%20is%20caused,I%20miss%2Dspell%20my%20ESSID.

It doesn't happen if you use the normal network manager controls to setup WiFi.

gearhead, 2024-01-02:
I appreciate the link, but already found it checked that. I have no stations listed in my wpa_supplicant.conf, so I cannot misspell one. Also, I set the WiFi region with raspi-config. Is there something else I might be missing?

ghollingworth, 2024-01-02:
Can you list the instructions for reproduction?  Start from:

1) Install a new lite image to SD card using RPi-Imager v 1.8.4

ezplanet, 2024-10-19:
Reporting same issue as above on RPI4, same logs, same sympthoms. Requesting to reopen this case or point me to one that might have been open in its place.

gearhead, 2025-07-03:
1) Install lite image
2) update
3) install wpa_supplicant
4) uninstall NetworkManager
5) enable systemd-networkd
6) disable wpa_supplicant
7) enable and start wpa_supplicant@wlan0
8) either edit /etc/wpa_supplicant/wpa_supplicant-wlan0 to add an ssid or use wpa_cli to add and enable it
9) turn off the ssid

If the ssid is 'known' but not visible, wpa_supplicant fills the log with the error message unless it is connected to that ssid or anoteh rknown ssid.

ghollingworth, 2025-07-03:
We only support using NetworkManager