RPi-Distro/repo #122: [RaspberryPi Zero] After a couple reboots, usb is unusable

agherzan, 2019-02-01:
### Setup

-  Pi power supply
-  PiZero (regular)
-  ThePiHut USB-OTG adapter + PiHut wifi dongle
-  wifi config added
-  enable_uart=1 added to config.txt
-  connect to the device over the UART

### Symptom
After a couple of reboots, usually 2-5, the wifi dongle doesn't expose a wifi interface anymore and `dmesg` reports USB errors as it follows:

```
pi@raspberrypi:~$ dmesg |grep -i usb
[    0.132776] usbcore: registered new interface driver usbfs
[    0.132922] usbcore: registered new interface driver hub
[    0.133154] usbcore: registered new device driver usb
[    0.356701] usbcore: registered new interface driver smsc95xx
[    0.390035] dwc_otg 20980000.usb: base=0xf0980000
[    0.842107] dwc_otg 20980000.usb: DWC OTG Controller
[    0.844754] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    0.847462] dwc_otg 20980000.usb: irq 56, io mem 0x00000000
[    0.855589] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    0.858255] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.860806] usb usb1: Product: DWC OTG Controller
[    0.863394] usb usb1: Manufacturer: Linux 4.14.79+ dwc_otg_hcd
[    0.865977] usb usb1: SerialNumber: 20980000.usb
[    0.869616] hub 1-0:1.0: USB hub found
[    0.876245] usbcore: registered new interface driver usb-storage
[    0.927663] usbcore: registered new interface driver usbhid
[    0.930255] usbhid: USB HID core driver
[    2.246421] usb 1-1: new full-speed USB device number 2 using dwc_otg
[    2.492954] usb 1-1: device descriptor read/64, error -71
[    2.832977] usb 1-1: device descriptor read/64, error -71
[    3.172930] usb 1-1: new full-speed USB device number 3 using dwc_otg
[    3.412559] usb 1-1: device descriptor read/64, error -71
[    3.752983] usb 1-1: device descriptor read/64, error -71
[    3.873157] usb usb1-port1: attempt power cycle
```

```
pi@raspberrypi:~$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

When in this state, only unplug/plug fixes the issue.

Tested this with both updated Raspbian Stretch and Yocto based distro. 

### Investigation

- the power supply doesn't seem to make a difference (USB cable or official power supply)
- switching Pi Hut dongle doesn't seem to make a difference (tried 3)
- Switching USB-OTG cable doesn't seem to make a difference (tried 2)
- switching PiZero doesn't seem to make a difference (tried 2)

XECDesign, 2019-02-01:
Since this is not really a packaging issue, it probably belongs in https://github.com/raspberrypi/linux/issues

agherzan, 2019-02-01:
I started from the distro to see if other people experienced this. I'll redirect it to the kernel repo.

closed 2019-02-01 by agherzan

floion, 2019-02-01:
@agherzan was this issue moved to a different repo?

agherzan, 2019-09-20:
https://github.com/raspberrypi/linux/issues/2858