RPi-Distro/repo #123: Add 64bit packages

TheCoder3156, 2019-02-01:
Would it be possible to add 64bit (aarch64) packages to the repo? It so people could modify https://github.com/RPi-Distro/pi-gen accordingly and make their own 64bit raspbian. This would also come with the advantage of the raspberry pi team not having to maintain 64bit images. If you cannot do this then can you please tell me how to make my own 64bit version of the repo? Thanks!

XECDesign, 2019-02-02:
No plans to do that right now. If you would like to do it, here's one way to go about it:

Set up debmirror to pull in all the source packages.

Set up reprepro

Set up sbuild

Write a script that will pass every source package to sbuild and import the result into reprepro.

Another handy feature of reprepro is the build-needed command, which will list every unbuilt source package, You can just import all the source packages from your debmirror directory to reprepro, and pass the unbuilt packages over to sbuild and pack to reprepro.

I think you'll find some packages will fail to build. Some will not have a source package and be .debs only. Others may build but have bugs that cause issues on 64bit systems. So it may take a day or two to sort everything out.

TheCoder3156, 2019-02-02:
Okay probably not doing that then. What are the packages used specifically for the pi to make it run faster/work at all? That way I could use Debian for aarch64 and just add those pi specific packages.

XECDesign, 2019-02-02:
The main packages to run at all are the kernel and the firmware. Debian should already have you covered - https://packages.debian.org/stretch/raspi3-firmware

Optimizations are a bit more difficult and you're mostly on your own there. Since I don't run aarch64 on my pi, I can't say much about it. Some of what we do is very simple - like enabling certain configure flags in some packages. Sometimes we'll have somebody spending weeks or months on a single package and most of that work is not currently usable on a 64bit system.

XECDesign, 2019-02-04:
We don't support Aarch64, so you're somewhat on your own here. You can check other 64bit distros and see how they perform. If they're better, you can check what they're doing. I believe you'll want to use the vc4 driver (should come with a recent enough kernel) and at least the Buster version of Mesa.

TheCoder3156, 2019-02-06:
Thank you for your time. Hopefully this will make it easier for others to use aarch64.

closed 2019-02-06 by TheCoder3156