RPi-Distro/repo #127: WiFi issues

XECDesign, 2019-02-23:
https://www.raspberrypi.org/forums/viewtopic.php?p=1433138#p1432916

dachshund-digital, 2019-04-11:
Can we get some response to this issue?  Experiencing issues on multiple Pi devices where disable of tls 1.0 and 1.1 per latest wpa_supplication package recommendations results in no DHCP ip address acquired.

XECDesign, 2019-04-12:
That message is only providing information, it's not a suggestion or recommendation. 99% of users should just ignore it. The linked thread doesn't seem to be related to this at all.

There's no response because there's a lot of information in that thread which I'd need to go over carefully and try to reproduce the problem. It would require a good chunk of time set aside.

dachshund-digital, 2019-04-12:
Thanks for the qualification.  IMHO, the wording and implication of the 'warning' is not just information in perception. Thus 99% of users will attempt to act in some manner, be it misguided or not, as you noted.  Thus, this only adds confusion and a negative perception.  At a minimum better communication should be done to explain this is more informational than not, if that is really the case.  Moreover, since RPi-Distro has a strong voice, it should use it, IMHO, and encourage the various parties to change or update information to provide better result.  Of course, this would only be done after due diligence as your noted.

brubbel, 2020-04-01:
Just to confirm: RPI4, 4.19.97-v7l+, Raspbian Lite
WiFi connection to (some) routers give a `wlan0: Lost carrier` every ~5 mins. A scan every 4 minutes seems to be a workaround.

dg1sek, 2020-06-06:
I seem to (?) have similar problems. 

I noted in other thread that it's not very clear whether the driver configuration should be

**/lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt**

or whether the correct driver is

**/lib/firmware/brcmo/brcmfmac43455-sdio.txt**

I will reinstall raspbian on my RPI4b - currently still on a very early version which didn't have the /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt file at all. 

I strongly suspect that these disconnects have something to do with a low level driver configuration issue and would start searching here - but I lack documentation around the brcm driver.

HBrydon, 2020-09-19:
Bump. This is Sep 2020 and I am still seeing the problem, and have seen it since my first Pi in 2012. Is this getting any attention?
It seems silly that we have to do workarounds for a stock system to keep the network going.

dg1sek, 2020-09-19:
I never concluded my analysis of the problem completely, but I'm pretty sure that one root cause is a conflict between "network manager" and "wicd" both trying to manage the Wi-Fi. Choose one and stick to that. Knowing that the UI of RPi seems to be using "network manager" disable everything else. Then check with `nmtui` if you can fix it. Sorry I can't help any further as I didn't have the time to dig completely into it.

Nodi-Rubrum, 2020-09-19:
Part of the issue seems to be related to E2000 routers.  I have 3, and 2 of the 3 drop as noted above.  Only the V3 (newest hardware) seems to be more stable.  But as good as the E2000 router were and are still, they are very old actually.

XECDesign, 2020-12-08:
Haven't managed to reproduce any issues. If anybody is still having trouble with the latest buster packages, please shout.

dachshund-digital, 2020-12-08:
Right now, with the latest updates, I am only seeing some odd disconnections for short periods, and most of the time on PiZero devices, they happen to be longest distance from APs as well.  I think I can safely qualify the short disconnects as common wireless interference.  Appear to be completely random as well, not like before, in any case.

closed 2020-12-22 by no-response[bot]