RPi-Distro/repo #130: python3-six is packaged incorrectly, breaking virtualenv

MatthewWilkes, 2019-03-21:
The package `python3-six` contains the following files:

    /.
    /usr
    /usr/lib
    /usr/lib/python3.5
    /usr/lib/python3.5/dist-packages
    /usr/lib/python3.5/dist-packages/six-1.12.0.egg-info
    /usr/lib/python3.5/dist-packages/six-1.12.0.egg-info/PKG-INFO
    /usr/lib/python3.5/dist-packages/six-1.12.0.egg-info/SOURCES.txt
    /usr/lib/python3.5/dist-packages/six-1.12.0.egg-info/dependency_links.txt
    /usr/lib/python3.5/dist-packages/six-1.12.0.egg-info/top_level.txt
    /usr/lib/python3.5/dist-packages/six.py
    /usr/share
    /usr/share/lintian
    /usr/share/lintian/overrides
    /usr/share/lintian/overrides/python3-six

Notice that the actual code is installed to `/usr/lib/python3.5`. Debian mainline has:

    /usr/lib/python3/dist-packages/six-1.8.0.egg-info/PKG-INFO
    /usr/lib/python3/dist-packages/six-1.8.0.egg-info/dependency_links.txt
    /usr/lib/python3/dist-packages/six-1.8.0.egg-info/top_level.txt
    /usr/lib/python3/dist-packages/six.py
    /usr/share/doc/python3-six/changelog.Debian.gz
    /usr/share/doc/python3-six/changelog.gz
    /usr/share/doc/python3-six/copyright

That is, the python3 dist packages directory is shared by minor versions.

While Python will accept this, virtualenv only accepts `pythonx.y/dist-packages` as a valid system path for python2. See https://github.com/pypa/virtualenv/blob/master/virtualenv_embedded/site.py#L261-L266

The upshot of this is that any packages that are installed through apt into `/usr/lib/python3/dist-packages` (such as `python3-traitlets`) are available in a virtualenv but those in  `/usr/lib/python3.5/dist-packages` (such as `python3-six`) will not be found.

To demonstrate this, install `python3-traitlets` and compare using the system python:

    pi@pvoutput:/tmp $ python3
    Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
    [GCC 6.3.0 20170124] on linux
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import traitlets
    >>> 

to a virtualenv:

    pi@pvoutput:/tmp $ virtualenv --python=python3 --system-site-packages testing
    Running virtualenv with interpreter /usr/bin/python3
    Using base prefix '/usr'
    New python executable in /tmp/testing/bin/python3
    Also creating executable in /tmp/testing/bin/python
    Installing setuptools, pip, wheel...
    done.
    pi@pvoutput:/tmp $ /tmp/testing/bin/python
    Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
    [GCC 6.3.0 20170124] on linux
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import traitlets
    Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    File "/usr/lib/python3/dist-packages/traitlets/__init__.py", line 1, in <module>
        from .traitlets import *
    File "/usr/lib/python3/dist-packages/traitlets/traitlets.py", line 57, in <module>
        import six
    ImportError: No module named 'six'
    >>> 

I recommend moving all files in `/usr/lib/python3.5/dist-packages` to `/usr/lib/python3/dist-packages` to solve this.

XECDesign, 2019-03-21:
@spl237 Looks like an issue with using py2deb

spl237, 2019-03-29:
Should be fixed in latest upload.

closed 2019-03-29 by spl237