RPi-Distro/repo #131: chromium crashes after today update

jmeile, 2019-03-22:
Hi

My Chromium kiosk was working before today's update. It was setup four days ago. The following packages were installed:
chromium-browser:armhf 65.0.3325.181-0+rpt4 72.0.3626.121-0+rpt3
chromium-codecs-ffmpeg-extra:armhf 65.0.3325.181-0+rpt4 72.0.3626.121-0+rpt3

I'm sure about this because minutes before the update, everything was working perfectly.

I have an nginx local web server, which loads a local php file. There I will rotate between different pages. The first page is normal html with a picture slider. After five minutes, a second page will be shown. There a html5 video should be played; however, instead of this, the following page will be shown:

""""
Aw, Snap!
Something went wrong while displaying this webpage.
"""

There is no error on the nginx log file nor the syslog. I was doing:
tail -f /var/log/nginx/error_log
tail -f /var/log/syslog

when testing and nothing was logged.

Here what I have:
Hardware: Raspberry Pi3B+

OS:
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2

Firmware:
Feb 12 2019 19:42:42
Copyright (c) 2012 Broadcom
version 8eff5e4023657a8b3b59e1f90dc966f62d74908c (clean) (release) (start)

Kernel:
Linux iwf-vitrine 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

Any way of reverting to the older version?

Here is what I had before according to: /var/log/apt/history.log
chromium-browser:armhf (65.0.3325.181-0+rpt4)
chromium-codecs-ffmpeg-extra:armhf (65.0.3325.181-0+rpt4, automatic)

Todays update looks like:
chromium-browser:armhf (65.0.3325.181-0+rpt4, 72.0.3626.121-0+rpt3)
chromium-codecs-ffmpeg-extra:armhf (65.0.3325.181-0+rpt4, 72.0.3626.121-0+rpt3)

I tried
apt-get install chromium-browser:armhf=65.0.3325.181-0+rpt4
apt-get install chromium-codecs-ffmpeg-extra:armhf=65.0.3325.181-0+rpt4

But the only thing I get is:
E: Version '65.0.3325.181-0+rpt4' for 'chromium-browser' was not found

By the way, there is nothing under:
/.config/chromium/Crash Reports

Best regards
Josef

XECDesign, 2019-03-22:
What would be a way to reproduce it? I've left it playing html5 videos for a while and am not seeing it. If you could roll up a testcase, it might be easier to figure out what goes wrong.

jmeile, 2019-03-22:
Now I tried:
```
apt purge chromium-browser

cd ~/.config
mv chromium chromium.bak
```

then reinstalled:
```
apt install chromium-browser
```

But there is no difference. It is still crashing. So, I guess it wasn't a plugin issue. I guess the error must be in the new video codes.

By the way, I had before only this plugin: [Cache Cleaner](https://chrome.google.com/webstore/detail/cache-cleaner/ebjidllpabafnaopkkkejbhaieciaink), but since I deleted the configuration it should be gone now.

jmeile, 2019-03-22:
> 
> 
> What would be a way to reproduce it? I've left it playing html5 videos for a while and am not seeing it. If you could roll up a testcase, it might be easier to figure out what goes wrong.

Ok, I reduce it to the following:

1. Setup a Raspberry-pi kiosk according to: [this instructions](https://die-antwort.eu/techblog/2017-12-setup-raspberry-pi-for-kiosk-mode)
2. On your **/etc/xdg/openbox/autostart** replace the url by: **http://localhost**
3. Setup an nginx local server.
4. put this index.php file on your default web root:
```
<html>
  <body>
    <video>
      <source src="video.webm"/>
    </video>
  </body>
</html>
```
5. Put also a video: video.webm there.
6. Now reboot your raspberry pi

You will be automatically logged in and the kiosk mode will start. Here chromium will chrash.

If I just drop that video tag, then it will work ok. I even tried with an mp4

jmeile, 2019-03-22:
Anyway of reverting to the previous chromium version? This is unfortunately a productive system and I can't debug on it. Right now I also don't have any other raspberry pi.

I guess I will reinstall everything on Monday and see if it works.

XECDesign, 2019-03-22:
Normally, it wouldn't be too difficult to downgrade. Annoyingly, the server didn't have enough room to keep both copies, so I had to remove the previous version and waybackmachine doesn't have it caches. Sec, I'll see if I can put the files somewhere else for you.

XECDesign, 2019-03-22:
Okay, I think these .debs should be enough: http://repo.xecdesign.com/chromium_old.tar.xz

jmeile, 2019-03-22:
The packages didn't work. Now I got:
```
ERROR
The requested URL could not be retreived

The following error was encountered while trying to retrieve the URL: http://localhost
  Access Denied
```

I rechecked the permissions and they are right. If I install the newer chromium version again, then it works.

Here is how I installed your files:
cd /tmp
mkdir chromium
cd chromium
tar xf chromium_old.tar.xz
dpkg -i *.deb

XECDesign, 2019-03-22:
Everything installed without errors? Not sure where to go from here then. I'm not aware of anything else being required.

jmeile, 2019-03-23:
Hi

There was just a warning telling me that I was downgrading, but it installed ok. Anyway, I will try again on Monday from a  fresh Raspian Lite and the latest chromium. If it fails there again, then it must be something wrong with this update.

Best regards
Josef

jmeile, 2019-03-25:
Ok, first I tried with the full desktop version of Raspbian (the recommended option with NOOBS), here the versions:
```
Distro: Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
Chromium: Version 72.0.3626.121 (Offizieller Build) Built on Raspbian , running on Raspbian 9.8 (32-Bit)
```

Here Chromium works ok with html5 videos. It won't crash.

Then I installed Raspbian Lite, and updated it. The distro and chrome version were exactly the same as above.

Next, I setup the Chromium-Kiosk as indicated on my previous posts. Then instead of installing nginx, I just put a link to a youtube video inside: **/etc/xdg/openbox/autostart**. When the kiosk starts, it will crash. This was working before the update from last friday (22.03.2019).

Then I just decided to removed the packages I installed for the kiosk:
`sudo apt purge xserver-xorg x11-xserver-utils xinit openbox chromium-browser`

Now: if you see the kiosk tutorial mentioned [before](https://die-antwort.eu/techblog/2017-12-setup-raspberry-pi-for-kiosk-mode), then you will see that the author installed the x11 environment and chromium as follows:
```
sudo apt install --no-install-recommends xserver-xorg x11-xserver-utils xinit openbox
sudo apt install --no-install-recommends chromium-browser
```

I just decided to remove: **"--no-install-recommends"** and install the packages again:
```
sudo apt install xserver-xorg x11-xserver-utils xinit openbox
sudo apt install chromium-browser
```

Finally I setup the kiosk again and you guessed it: **It worked**.

So, I guess the last update added a hidden dependency to Chromium. For me this seems to be a bug, but since I don't have much time for debugging, I will leave the kiosk with the full packages. This is actually not that bad.

Thanks anyways for your support.

Best regards
Josef

XECDesign, 2019-03-25:
Thanks for looking into it, that actually gives me some ideas on what the problem may be. I'll dig in when I have a spare moment.

XECDesign, 2019-03-26:
I think the important package is `libgl1-mesa-dri`

jmeile, 2019-03-27:
Yes, I can confirm that after running:
`apt install --no-install-recommends libgl1-mesa-dri `

Chromium won't crash. Here the output of that command:
The following additional packages will be installed:
```
  libdrm-freedreno1 libllvm3.9 libsensors4
Suggested packages:
  lm-sensors
Recommended packages:
  libtxc-dxtn-s2tc | libtxc-dxtn-s2tc0 | libtxc-dxtn0
The following NEW packages will be installed:
  libdrm-freedreno1 libgl1-mesa-dri libllvm3.9 libsensors4
```

So, I guess Chromium in deed needs the packages:
```
libdrm-freedreno1
libgl1-mesa-dri
libllvm3.9
libsensors4
```

for rendering html5 video.

XECDesign, 2019-03-27:
Added `libgl1-mesa-dri` as a dependency.

closed 2019-03-27 by XECDesign

kevin-bache, 2019-04-02:
I'm having the same issue with a chromium kiosk. This does not seem to be fixing it though. Anyways to revert to a previous version? Chromium 65 to be exact?

jmeile, 2019-04-02:
On my case, the issue was coming when first opening a page with an html5 video. The issue went away with the next update (or at least I thought it).

Now, it is appearing after about 5 hours of playing the same videos in a loop together with some image sliders. I just removed one of the videos today and it seems to run without any problem.

I will add it tomorrow and see if the issue reappears.

Is there any log file for Chromium?

kevin-bache, 2019-04-02:
@jmeile sounds like we are working on similar projects. You can enable logging if you launch chromium with this switch

`--enable-logging=stderr
`

jmeile, 2019-04-03:
@kevin-bache Yes, I'm also doing a chromium kiosk with a Raspberry-pi. Thanks for the hint to the command line switch.

By the way: today I activated again the video that I thought that was causing the problem. Chromium didn't crash.

How is it by you? Is it crashing a lot? I even found some instructions that tell you to disable some things:

https://windowsreport.com/aw-snap-google-chrome-windows-10

Some of them don't make sense for a kiosk, but I guess you could try others, ie: 

- Disable Predict network actions to improve page load performance
- Disable your extensions
- Clear browsing data -> I already do this before starting chromium. I don't have the system here, but I think is under: ~/cache/chromium
- Disable Sandbox mode

I even tried something there: "Disable hardware acceleration", but the problem just came more often.

Best regards
Josef

jmeile, 2019-04-04:
Today it crashed again. I will see if using just webM videos solves the issue.

XECDesign, 2019-04-04:
Are you running it with any special flags? How much GPU memory do you have available? Are you using the legacy graphics stack, fake KMS or full KMS?

jmeile, 2019-04-05:
Hi

Chromium became unstable as I added more videos to play. It will crash after a while when loading a video.

Here the options I'm using on boot/config.txt
```
disable_overscan=1
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=82
dtparam=i2c_arm=off
dtparam=spi=off
dtparam=audio=on
hdmi_force_hotplug=1
enable_uart=0
#gpu_mem=128
start_x=0
dtoverlay=vc4-kms-v3d
```

Then this is what I set with raspi-config:
```
Predictable network interface names -> Disabled
Console Autologin -> Set
Wait for Network at Boot -> Enabled
Camera -> Disabled
SSH -> Enabled
VNC -> Disabled
SPI -> Disabled
I2C -> Disabled
Serial -> Disabled
1-Wire -> Disabled
Remote GPIO -> Disabled
Overclock -> Disabled
Expand Filesystem -> Not supported due partition layout
Overscan -> Disabled
Memory split: 128
Audio through HDMI or 3.5mm jack -> Auto
Resolution -> DMT Mode 82 1920x1080 60Hz 16:9
Pixel doubling -> Disabled
GL Driver -> OpenGL desktop driver with full KMS
```

This is what I have under: /home/pi/.bash_profile:
`[[ -z $DISPLAY && $XDG_VTNR -eq 1 ]] && startx -- && unclutter -idle 0.01 -root`

Here the contents of /etc/xdg/openbox/autostart (where the chomium kiosk is set):
```
# Disable any form of screen saver / screen blanking / power management
xset s off
xset s noblank
xset -dpms

# Allow quitting the X server with CTRL-ATL-Backspace
setxkbmap -option terminate:ctrl_alt_bksp

# Deletes Chromium cache on startup
rm -Rf ~/.cache/chromium

# Start Chromium in kiosk mode
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' ~/.config/chromium/'Local State'
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/; s/"exit_type":"[^"]\+"/"exit_type":"Normal"/' ~/.config/chromium/Default/Preferences

chromium-browser --disable-infobars --kiosk 'http://localhost'
```

By the way: where are the chromium logs stored? According to the docs, the options "--enable-logging=stderr --v=1" should save "chrome_debug.log" in Chrome's user data directory, but I don't find it under: "/home/pi/.config/chromium".

Now I'm trying webm videos, but the same video will be always shown. All videos are named equal: video.webm, but they are located in different folders, so, chromium should try them as different videos because they have different urls. I don't think that this is a cache issue. I always delete cache on start.

Anyway, I will be away next week, so, as soon as I come, I will do another small test case. As I said: I'm using image and text sliders where also different videos will be shown according to the current text slider.

Best regards
Josef

XECDesign, 2019-04-05:
For video playback, the fkms and kms drivers are not as fast. Do you experience issues with a good chunk of gpu mem, the legacy driver and `--disable-gpu-compositing` ?

jmeile, 2019-04-05:
I will try that after next week.

davidmmgo, 2019-06-25:
Hi!
I'm having the same exact issue. 
I have a kiosk setup, running chromium on a RaspberryPi (Raspbian). It simply plays local mp4 videos on an html page. It crashes after a few videos (around 40) - doesn't matter if they're always the same video, and entirely different ones.
Our initial setup was stable, running Chromium 65 on Raspbian Lite 9.3. After updating, I was never able to rollback to that exact configuration.

My test script:
`window.onload = function () {

    const videoUrl = 'myvideo.mp4';
    const videoTag = document.getElementById('video_place');
    var attemptCount = 0;

    function logMessage(msg) {
        console.log(msg + ' #' + attemptCount);
    }

    function playVideo() {
        ++attemptCount;
        logMessage('NEW ATTEMPT');
        videoTag.src = videoUrl;
    }

    videoTag.oncanplay = function () {
        logMessage('CAN PLAY');

        videoTag.play()
            .then(() => { logMessage('SUCCESS IN PROMISE'); })
            .catch(() => { logMessage('ERROR IN PROMISE'); });
    };
    videoTag.onended = function () {
        logMessage('ENDED');
        playVideo();
    };

    playVideo();
};`

Using chromium's logging, I get the following (not exactly sure what they mean though): 
WARNING:mmal_video_decoder.cc(1812) ### Mmal reports error 3
INFO:mmal_video_decoder.cc(944) --- In flight low: 3/8

After days of testing and googling, I've come across this thread.
@jmeile Did you manage to get yours working stably? If so, I'm very interested in knowing what your findings were, and what your current configuration is.

Thank you,
David

jmeile, 2019-06-26:
Dear David

Unfortunately, my workarround was disabling the video playback. I was still getting the same issue and since I only have a Rapsberry-pi, I can't debug this intensively.

I believe that this may be related to the garbage collector of Chromium. I guess it is not cleaning its memory properly and that's why it happens. I think that you may solve this by restarting the x server on a regular basis. I would start trying it each 8 hours and see if your problem is gone. This is the command I'm use for doing this:

`sudo kill -9 `ps -t tty1 | awk {'print $1'} | grep -v PID`

On my kiosk mode, it will kill x and then restart it.

I was also using a script to monitor my kiosk from my office called take_screenshot (see attachment). Just put the following entry on the crontab of the user running your kiosk (ie: pi user)

`00,05,10,15,20,25,30,35,40,45,50,55 * * * * /home/pi/take_screenshot`

Please note the following:
- I'm using jpeg images with a quality of 50 and a resolution of: 1920x1080 (FullHD).
- My Chromium is running in full screen without any status bar.

Based on that, I figured out that the image been captured as chromium crashes, error:
```
Aw, Snap!
Something went wrong while displaying this webpage.
```

has the size of: 36694 bytes. So, I just compared if the generated screenshot has this size and so, then the x server will be restarted. If you want to use a different quality for your screenshots, then you will have to comment first the last lines of my script:
```
if [ "$FILE_SIZE" -eq "$CHROMIUM_ERROR_SIZE" ] ; then
  echo "Restarting x on $DAY.$MONTH.$YEAR at $HOUR:$MIN:$SEC" | tee -a $LOG_FILE
  /home/pi/startx
else
  echo "X wasn't restarted"
fi
```

Then wait until the error happens and see what's its size. After that, change the value of: CHROMIUM_ERROR_SIZE to fit your requirements.

Other thing you need to change is the value of: "LOG_FILE" and "SCREENSHOT_DIR". On my pi, I'm mounting a samba share automatically, then I store all my screenshots there. By doing this, I'm able to watch the screenshots from my office by mounting that share on my laptop.

This is only a workarround. Not a solution. It seems that Chromium is broken.

Best regards
Josef

[take_screenshot.txt](https://github.com/RPi-Distro/repo/files/3329872/take_screenshot.txt)

davidmmgo, 2019-06-26:
Hi @jmeile , thanks for your reply.

We have a scheduled maintenance reboot every 24 hours. However, with short videos, we get crashes in about 10 minutes (roughly 40 video playbacks), regardless of how many unique videos there are (right now, we're testing with just one video).

I tried forcing the garbage collector by enabling and exposing it through the V8 engine and calling gc in my script, but it crashed all the same. I agree that it seems to be a memory related problem, but I suspect it might be from the video decoder itself. However, it's not that the Pi is running out of memory, from my observations with htop. 
When it crashes, I can see the last message it sends out is "Bus error". Do you also get that?

Since we still have a stable setup running (Chromium 65.0.3255.181 and Raspbian 9.3), built way before the update, we might end up creating an image from that and cloning it for other future deployments, since there's really no way of rolling back.
I'm sorry to hear that you had to remove one of the features of your product. Since our project's main focus is to display video in a kiosk-y way, I'm hoping that we'll be able to find and solve this issue.

Cheers, 
David

jmeile, 2019-06-26:
Hi David

Right now I'm on vacations, but I will see what is the message when Chromium crashes. I didn't include a log file here since I wasn't able to enable it on chromium. I will try it next time.

Best regards
Josef

davidmmgo, 2019-06-26:
Hey Josef,

In order to enable logging, I launch Chromium using `--enable-logging=stderr` via the terminal, and that's where I see the error message.
I'm looking forward to your feedback.
In the meantime, enjoy your vacations!

Cheers, 
David

chengpangboon, 2019-07-26:
> Added `libgl1-mesa-dri` as a dependency.

Facing the same problem here when playing HTML5 video. Hopefully there will be solution soon, or possible to downgrade to 65.

UshakovVasilii, 2019-12-14:
I had chrome crashes with several video loop on 74 version. 78 version from untested repo works fine!

mathieukergosien, 2019-12-18:
> 
> 
> I had chrome crashes with several video loop on 74 version. 78 version from untested repo works fine!

Hello,
How did you do that? In the testing branch of the apt sources of raspbian, I still have the version 74...