RPi-Distro/repo #132: alsa-lib: please add card conf for rpi-wm8804 driver in kernel 4.19

HiassofT, 2019-03-25:
The various WM8804 based soundcard drivers (Allo, IQaudio, JustBoom, Hifiberry digi cards) were refactored into a single driver in kernel 4.18/19.

As the ALSA card conf files are tied to the driver name this requires a new file to get the iec958 PCM - which is required by Kodi to enable AC3/DTS passthrough - on kernel 4.19. See eg this issue on the kodi forums https://forum.kodi.tv/showthread.php?tid=342262

In LibreELEC we use this file https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/filesystem/usr/share/alsa/cards/RPi-WM8804.conf which will work for Debian/Raspbian as well

XECDesign, 2019-03-25:
Safe to assume adding it won't cause issues for people on 4.14?

HiassofT, 2019-03-25:
yes, it'll be safe for 4.14

XECDesign, 2019-03-25:
Updated version should be available in the repo now.

HiassofT, 2019-03-25:
thanks a lot!

closed 2019-03-25 by HiassofT