RPi-Distro/repo #133: How do I add a program to apt

WitherTech, 2019-03-26:
I made a program. How do I make it so people can install it with apt?

XECDesign, 2019-03-26:
We don't normally take third party packages like that, but if it's something that would benefit a significant portion of raspberry pi users, send an email to serge at raspberrypi.org

WitherTech, 2019-03-26:
What’s his email?

XECDesign, 2019-03-26:
That was the email. I didn't use the `@` symbol to avoid getting too much more spam.

WitherTech, 2019-03-26:
Ok

WitherTech, 2019-03-27:
He said it does not meet Debian standards
What are they?

XECDesign, 2019-03-27:
Like I said in the email, it should pass piuparts and lintian. You can't assume a network connection is available. You can't git clone from postinst. It needs to be a proper source package with a dsc file. But even if it was packaged perfectly, I don't know that we can include it. The recommendation is to sort out packaging and try to get it into Debian instead.

closed 2019-06-05 by XECDesign