RPi-Distro/repo #135: Chromium kiosk crashes after updating to 72

kevin-bache, 2019-04-02:
Chrome kiosk crashes after the update. Was working perfectly before version 65 running on raspi 3+

I get this error

WARNING:software_output_device_x11.cc(105)] Display attributes incompatible with SHM

XECDesign, 2019-04-03:
Please provide instructions on how to reproduce, starting from the latest raspbian desktop image (updated and rebooted).

kevin-bache, 2019-04-03:
Hi @XECDesign. It's a Chromium digital signage app running in docker containers. 

It's a tv app. The app has two containers;

**BACKEND** - which is a node/react app communicating via websockets.

**FRONTEND** - chromium browser connects to the exposed port.

We are running on the Raspberry Pi 3+. OS images are;

**BACKEND**: - FROM resin/raspberrypi3-node:8

**FRONTEND** - FROM balenalib/raspberrypi3-debian:stretch

As it stands. I have;

**entrpoint.sh**

```
#!/bin/bash
umount /dev/shm && mount -t tmpfs shm /dev/shm

env > cron_env_file

# start cron
cron

sleep 25
xinit /home/viewer/launchBrowser.sh -- -nocursor
```
**launchBrowser.sh**

```
#!/bin/bash
#Disable DPMS / Screen blanking
xset -dpms
xset s off
xset s noblank

URL=$URL
default='http://localhost:8888/'

while :        # run forever
do
    echo "Launching Browser"
    echo "Browser is ${URL}"
    DISPLAY=:0 sudo chromium-browser --no-sandbox --window-size=7000,7000 --kiosk --start-fullscreen -a --disable-infobars --no-first-run --disable-web-security --user-data-dir --incognito --allow-file-access-from-files --enable-offline-auto-reload-visible-only --disable-gpu --force-device-scale-factor=0.5 ${URL:-$default} viewer
done
```
**Dockerfile:**

```
FROM balenalib/raspberrypi3-debian:stretch

USER root

RUN apt-get update && apt-get install -y --no-install-recommends \
lsb-release \
nano \
chromium-browser \
omxplayer \
scrot \
cron \
sed \
xorg \
xinit \
xinput \
xserver-xorg \
xserver-xorg-legacy \
xserver-xorg-input-evdev \
xserver-xorg-input-libinput \
x11-xserver-utils && \
apt-get clean && rm -rf /var/lib/apt/lists/*

RUN sed -i "s|allowed_users=console|allowed_users=anybody|" /etc/X11/Xwrapper.config

RUN groupmod -g 19 input # may differ

RUN groupadd viewer && \
echo ‘notsecure\nnotsecure’ | adduser --home /home/viewer --ingroup viewer viewer && \
usermod -aG tty viewer && \
usermod -aG input viewer

COPY entrypoint.sh /home/viewer/entrypoint.sh
COPY launchBrowser.sh /home/viewer/launchBrowser.sh

COPY screenshoter.sh /home/viewer/screenshoter.sh
COPY restartBrowser.sh /home/viewer/restartBrowser.sh
COPY restartAllServices.sh /home/viewer/restartAllServices.sh
COPY cron_env_file /home/viewer/cron_env_file

RUN chmod +x /home/viewer/*
RUN chown viewer:viewer /home/viewer/*

# Add crontab file in the cron directory
ADD crontab /etc/cron.d/hello-cron
# Give execution rights on the cron job
RUN chmod 0644 /etc/cron.d/hello-cron
# Apply cron job
RUN crontab /etc/cron.d/hello-cron

# Allow viewer user to start cron daemon with sudo
RUN echo 'viewer ALL=NOPASSWD: /usr/sbin/cron' >>/etc/sudoers

CMD bash -C "/home/viewer/entrypoint.sh";"bash"
```

kevin-bache, 2019-04-03:
To reproduce for reference though it seems any youtube page will crash with the "aw snap" if it's run with "--disable-gpu" flag

If I run our app without that flag it won't crash but stalls and has other performance issues. This all worked flawlessly in 65 so I'm wondering what changed and how I can configure this. I'd really rather not rebuild an older version of Chromium

XECDesign, 2019-04-03:
> To reproduce for reference though it seems any youtube page will crash with the "aw snap" if it's run with "--disable-gpu" flag

Thanks, I'll pass it on to the dev.

Do you still have performance issues if you run without `--disable-gpu`, but with `--disable-gpu-compositing`?

kevin-bache, 2019-04-03:
Hi @XECDesign hmm --disable-gpu-compositing doesn't have "aw snap" errors but the videos don't play.

XECDesign, 2019-04-03:
Strange. We ship with ` --disable-gpu-compositing` and it has been fine so far. Sounds like a bad interaction with something else in your setup.

Anyway, sounds like `--disable-gpu` should be a good hint for where to look.

kevin-bache, 2019-04-03:
Hi @XECDesign I actually got it working by remove `--disable-gpu` and increasing video memory to 128MB

closed 2019-04-03 by kevin-bache

mboehm21, 2019-05-24:
Hi @kevin-bache we have the same issue. Could you please post your full Chromium settings? Thanks a lot.