RPi-Distro/repo #137: udisksadm does not respect "disk" group

lukehutch, 2019-06-01:
Users (in particular the `pi` user) should be able to mount or unmount disks without supplying a password using `udisksctl [un]mount -b /dev/sda1` etc., as long as they are in the `disk` group, using `sudo usermod -a -G disk pi`. (Note that the block devices `/dev/sdXN` are in the `disk` group already.)

However, after adding the user `pi` to the `disk` group and rebooting, `udisksctl` still requires a password unless you run it with `sudo`. 

I think this means that Polkit is not set up correctly on Raspbian.

lukehutch, 2019-06-02:
I noticed that the `pi` user is part of the `adm` group too, but it seems that this group has not been set up as an administrator whose password is not needed for privileged actions.

To try to override this I tried putting in `/etc/polkit-1/localauthority.conf.d/61-localauthority-override.conf` the following, which did not work:

```
[Configuration]
AdminIdentities=unix-user:0;unix-group:sudo;unix-group:adm
```

The goal was to override `/etc/polkit-1/localauthority.conf.d/50-localauthority.conf`:

```
# Configuration file for the PolicyKit Local Authority.
#
# DO NOT EDIT THIS FILE, it will be overwritten on update.
#
# See the pklocalauthority(8) man page for more information
# about configuring the Local Authority.
#

[Configuration]
AdminIdentities=unix-user:0
```

I then tried adding the following to `/etc/polkit-1/localauthority/90-mandatory.d/91-udisks2.pkla` to try to allow `udisksctl` actions to be run without any authorization, and this also did not work:

```
[udisks2]
Identity=uni-group:adm
Action=org.freedesktop.udisks2.*
ResultAny=yes
ResultInactive=yes
ResultActive=yes
```

According to my reading of the Polkit documentation (e.g. `man pklocalauthority`), both of the above should work. Am I doing something wrong, or is Polkit misconfigured?

artmg, 2021-07-06:
Although @lukehutch proposed the `disk` group, the [list of different system groups on Debian](https://wiki.debian.org/SystemGroups) states that is to permit raw disk access. Instead the Debian group description closest to this use case sounds like **plugdev**, which allows members to (un-)mount removable devices.

I use the folder 20-org.d as I would apply this authority across in my organisation, but for the OS, 10-vendor.d seems most apt [part of the structure](https://www.freedesktop.org/software/polkit/docs/0.105/pklocalauthority.8.html). Paste the following commands into the terminal to create the authority file: `/var/lib/polkit-1/localauthority/10-vendor.d/21-udisks2-plugdev-mount-other.pkla`. I chose just the [udisks2 **mount other seat** polkit action](http://storaged.org/doc/udisks2-api/2.7.5/udisks-polkit-actions.html) – you might consider others appropriate, but udisks2.* seems for too wide ranging for the description of the plugdev group, and for the use-case stated in the issue.

```
sudo tee /var/lib/polkit-1/localauthority/10-vendor.d/21-udisks2-plugdev-mount-other.pkla <<EndOfPkla
# if the user is in the Debian SystemGroup plugdev, which allows members to (un-)mount removable devices,
# then allow them to mount drives other than those internally seated, such as USB connected drive

[Allow plugdevs to mount other drives]
Identity=unix-group:plugdev
Action=org.freedesktop.udisks2.filesystem-mount-other-seat
ResultAny=yes
EndOfPkla
```

This is tested as working correctly in buster armv7l on Pi 3B

```
$ udisksctl mount -b /dev/sdb1
Mounted /dev/sdb1 at /media/pi/SDCARD.
$ udisksctl unmount -b /dev/sdb1
Unmounted /dev/sdb1.
```

XECDesign, 2021-07-06:
@spl237 I just had a look at this and from what I gather, you've added `/var/lib/polkit-1/localauthority/10-vendor.d/55-storage.pkla` to `raspberrypi-ui-mods`, which handles this. I'm guessing op is running the lite version, so they can't use that package without dragging in a lot of dependencies they don't need.

In Debian, there's gnome-initial-setup, which has equivalent functionality and in Ubuntu there's policykit-desktop-privileges. In all cases, the implication is that this is functionality intended for desktop environments only, but that's probably not a valid assumption for Raspberry Pi OS.

@spl237 Mind if I move the file into raspberrypi-sys-mods instead? On your end ui-mods would need to remove the file and add 'Depends: raspberrypi-sys-mods (>=20210706) '

spl237, 2021-07-06:
@XECDesign Yes, no problem with that - fits better in sys-mods anyway. I'll build a new package this afternoon with it removed.

XECDesign, 2021-07-07:
Should be fixed

artmg, 2021-07-09:
Yes, @XECDesign I was running Lite (although I can't comment for OP). In the absence of Desktop UI the udisksctl mount CLI method IS indeed a handy mechanism and your approach looks very sensible. Should I offer to test, when packages become available, or do you have team members for that?
Many thanks for dealing with this promptly

XECDesign, 2021-07-09:
The package is already in the repo, so an apt update and upgrade should be all that's needed.

https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/59fd502bd4fcf8f5f646ba6c7ad59c8756c3094e

artmg, 2021-07-09:
Yep, the file is now in there.

However, when I insert a USB storage device and attempt, e.g. `udisksctl mount -b /dev/sdb2`, I still get 

```
==== AUTHENTICATING FOR org.freedesktop.udisks2.filesystem-mount-other-seat ===
Authentication is required to mount Generic STORAGE DEVICE (/dev/sdb2)
Authenticating as: ,,, (pi)
Password:
```
meaning **`filesystem-mount-other-seat`** is the action needing allowing, as well as the 'filesystem-mount' you have in the 55-storage authority file. I'm not enough of a udisks expert to know the implications (and therefore risks) of 'other seat' devices, but if they include USB devices then they may well need authorising here too. 

Would you say it's reasonable to add that action into the list in that file?

artmg, 2021-07-21:
@XECDesign apologies if my response was not clear on the fact that Tests on your latest drop show it FAILS to clear the issue. The `filesystem-mount-other-seat` action needs adding to the allow list, as well as the `filesystem-mount`. I did not raise a PR as I'm not sure you accept them from amateurs, but can if you wish. Thanks

XECDesign, 2021-07-21:
I just haven't had time to confirm that or to check why `filesystem-mount-other-seat` would be needed on the lite image but not on the desktop. It's still on the todo list, just not the highest priority right now.

XECDesign, 2022-01-19:
I've just tried this on the latest lite image (after upgrading and installing udisks2) and it seems to just work. Is there a case that currently fails?

It seems to require authentication if you do it over ssh, which I guess is what `filesystem-mount-other-seat` is for?

artmg, 2022-01-23:
I carried out the following test case

* installed image **2021-10-30-raspios-bullseye-armhf-lite.zip**
* `sudo apt dist-upgrade`
* `sudo apt install udisks2`
	* this also installed: eject exfatprogs gdisk libatasmart4 libblockdev-crypto2 libblockdev-fs2 libblockdev-loop2 libblockdev-part-err2 libblockdev-part2 libblockdev-swap2 libblockdev-utils2 libblockdev2 libgpgme11 libgudev-1.0-0 libnspr4 libnss3 libparted-fs-resize0 libudisks2-0 libvolume-key1
* `groups | grep plugdev || groups`
	* first check you are in the 'plugdev' group
* `ls /dev/disk/by-id -la`
* `lsblk`
* `udisksctl mount -b /dev/sda1`

```
==== AUTHENTICATING FOR org.freedesktop.udisks2.filesystem-mount-other-seat ===
Authentication is required to mount Generic STORAGE DEVICE (/dev/sda1)
Authenticating as: ,,, (pi)
Password:
==== AUTHENTICATION COMPLETE ===
Mounted /dev/sda1 at /media/pi/MyVolLabel
```

* `ls -la /media/pi`

```
drwxr-x---+ 3 root root 4096 Jan 23 13:06 .
drwxr-xr-x  3 root root 4096 Jan 23 13:06 ..
drwxr-xr-x  3 pi   pi   3584 Jan  1  1970 MyVolLabel
```

So yes it works in the user context, so the mount point is created with the correct user permissions. However, as you say, it DOES still ask for user authentication, which is not really ideal  :(

Please let me know if any other tests or diagnostics might help.

closed 2022-02-02 by no-response[bot]