RPi-Distro/repo #139: sonic pi downgrade in buster.

plugwash, 2019-06-26:
Sam Aron has complained that the version of sonic-pi shipped in the raspberry pi foundation raspbian buster images is singificantly older than the version shipped in the raspberry pi foundation raspbian stretch images.

My research so-far has revealed

1. There is no sonic-pi package in the raspberry pi foundation's repository for raspbian buster.
2. There is a sonic-pi package in the plain raspbian buster repository, this is (modulo a rebuild) the same as what Debian ships and is rather outdated.
3. The "1:3.0.1+1" package in the raspberry pi foundation stretch repository cannot be installed on buster because of library dependencies.
4. There does not seem to be a corresponding source package for said package, so I cannot simply rebuild it.

I am currently trying to see if I can produce a 3.1.0 package based on debian's 2.10.0 packaging.

XECDesign, 2019-06-27:
I don't understand the complaint. The previous builds were provided by Sam.

plugwash, 2019-06-27:
> I don't understand the complaint. 

AIUI the complaint is that the raspberry pi foundation buster setup (not sure if it's in the image or reccomended software, but either way) shipped with a much older sonic-pi than the raspberry pi foundation stretch setup and the first sam was made aware of this was when users started using the image.

> The previous builds were provided by Sam.

It's really best to get away from shipping random binaries built god knows how if at all possible, A proper source package makes it much easier to rebuild/tweak for things like library name changes.

ukscone, 2019-06-27:
i got it down to one problem package in the buster repos.

sonic pi asks for  libqt5scintilla2-12v5 but in buster the nearest package is called  libqscintilla2-qt5-13 so not only version but also name problems. everything else seems to be fine, at least they installed. i'm guessing either Sam needs to supply a source tarball or at least provide a binary deb

XECDesign, 2019-06-27:
Yes, I was working on a proper source package quite a while back, but met some resistance which halted that effort.

As things stand, we are happy to put up an updated version of sonic pi. If Sam wants to provide an updated version, we'll gladly accept it. If somebody wants to package it up properly, even better. Until then, we have to make do with what we have.

spl237, 2019-06-27:
> It's really best to get away from shipping random binaries built god knows how if at all possible, A proper source package makes it much easier to rebuild/tweak for things like library name changes.

This. Sam has never been able to supply a proper source package - it took long enough to get him to supply a hacked-together deb package in spite of repeated requests. The build process for Sonic Pi is a mess - numerous sub-builds, bits being pulled in from other directories etc. Sam has never bothered to sort this out, and now it's apparently our fault and up to us to pay him to make it work. (Which he never did when we *were* paying him, in spite of requests to do so.)

We're not going to do anything about this - if Sam wants to either a) provide a Buster compatible deb or b) (preferred) make a proper source package that we can actually rebuild, then we can work with that, but for now, this is Sam's fault, and Sam's problem, and he can deal with it or live with things the way they are.

closed 2019-06-27 by spl237

spl237, 2019-06-27:
> It's really best to get away from shipping random binaries built god knows how if at all possible, A proper source package makes it much easier to rebuild/tweak for things like library name changes.

This. Sam has never supplied a proper Debian source package, in spite of repeated requests to do so. Probably because the build process for Sonic Pi is a mess - numerous sub-builds, bits being pulled in from other directories etc.

We're not going to do anything about this - if Sam wants to either a) provide a Buster compatible deb or b) (even better) make a proper source package that we can actually rebuild, then we can work with that. But in the absence of that, there's nothing we can do.

skade, 2019-06-27:
Replicating my comments from elsewhere.

I find the answers very sad. "upstream should provide a package for us" is always a bad way to frame things. I'm pretty sure the outlined issues can be fixed by combining forces into a concerted fundraiser instead of blaming the author and actually funding the time needed to better the packaging process.

Also, I'm confused why this issue is closed, the newest Raspberry Pi downgrading a major package seems like an issue to me, reasons or not.

spl237, 2019-06-27:
Raspberry Pi (who are the owners of the repo against which this issue has been raised) are not responsible for maintaining and updating every package in the image - we simply do not have the resource to do so.

Sonic Pi is a third-party package; it is the responsibility of the third party in question to maintain it (as is the case for every other third-party package we include) - the correct place to raise this issue is therefore on the Sonic Pi repo, not ours.

We have offered to host an updated package on our repo; failing that, we will include the upstream package for as long as it is compatible. Anything else is the responsibility of the package author and not that of Raspberry Pi.

skade, 2019-06-27:
I have not asked for maintenance, it seems like it's word against word who is responsible here and who has not well communicated with whom.

SonicPi is definitely a popular package, so I'd expect some more then just arguing mechanics?