RPi-Distro/repo #142: FreeCad SIGSEV at creating a new file

schorsch1976, 2019-07-05:
Dear maintainer,

i get a SIGSEV as i create a new file in freecad. It happens every time 
on my device. As this is a raspberry pi 4 with raspian buster, i asked 
in the forums where to report this bug and they pointed me to debian. Debian closed it. So it might be right here.

Here is the callstack

pi@pi4:~ $ gdb freecad
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later 
<http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from freecad...(no debugging symbols found)...done.
(gdb) start
Temporary breakpoint 1 at 0x1284c
Starting program: /usr/bin/freecad
[Thread debugging using libthread_db enabled]
Using host libthread_db library 
"/lib/arm-linux-gnueabihf/libthread_db.so.1".

Temporary breakpoint 1, 0x0001284c in main ()
(gdb) c
Continuing.
FreeCAD 0.18, Libs: 0.18R
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[New Thread 0xaf479210 (LWP 8240)]
libEGL warning: DRI2: failed to authenticate
[New Thread 0xaa883210 (LWP 8241)]
[New Thread 0xaa082210 (LWP 8242)]
[New Thread 0xa94ff210 (LWP 8243)]
[New Thread 0xa8cfe210 (LWP 8244)]
[New Thread 0xa7f38210 (LWP 8245)]
[New Thread 0xa2248210 (LWP 8246)]
[New Thread 0x9fdff210 (LWP 8247)]
Coin warning in cc_glglue_instance(): Error when setting up the GL 
context. This can happen if there is no current context, or if the 
context has been set up incorrectly.

Thread 1 "freecad" received signal SIGSEGV, Segmentation fault.
0xb3831e74 in XDefaultScreenOfDisplay () from 
/usr/lib/arm-linux-gnueabihf/libX11.so.6
(gdb) bt
#0  0xb3831e74 in XDefaultScreenOfDisplay () from 
/usr/lib/arm-linux-gnueabihf/libX11.so.6
#1  0xb5077ad0 in ?? () from /usr/lib/arm-linux-gnueabihf/libCoin.so.80c
#2  0xb5078454 in glxglue_init () from 
/usr/lib/arm-linux-gnueabihf/libCoin.so.80c
#3  0xb50710e0 in cc_glglue_instance () from 
/usr/lib/arm-linux-gnueabihf/libCoin.so.80c
#4  0xb4f22120 in SoGLRenderActionP::isDirectRendering(SoState const*) 
const () from /usr/lib/arm-linux-gnueabihf/libCoin.so.80c
#5  0xb4f24358 in SoGLRenderActionP::render(SoNode*) () from 
/usr/lib/arm-linux-gnueabihf/libCoin.so.80c
#6  0xb4f245a0 in SoGLRenderAction::beginTraversal(SoNode*) () from 
/usr/lib/arm-linux-gnueabihf/libCoin.so.80c
#7  0xb4f1e2fc in SoAction::apply(SoNode*) () from 
/usr/lib/arm-linux-gnueabihf/libCoin.so.80c
#8  0xb634b2f4 in Gui::SoBoxSelectionRenderAction::apply(SoNode*) () 
from /usr/lib/freecad-python2/lib/libFreeCADGui.so
#9  0xb63e5b64 in Gui::View3DInventorViewer::renderScene() () from 
/usr/lib/freecad-python2/lib/libFreeCADGui.so
#10 0xb63b6e34 in 
SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) () from 
/usr/lib/freecad-python2/lib/libFreeCADGui.so
#11 0xb63be598 in 
SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) () 
from /usr/lib/freecad-python2/lib/libFreeCADGui.so
#12 0xb438f590 in QWidget::event(QEvent*) () from 
/usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
#13 0xb44381bc in QFrame::event(QEvent*) () from 
/usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
#14 0xb4695118 in QGraphicsView::viewportEvent(QEvent*) () from 
/usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
#15 0xb63b7028 in 
SIM::Coin3D::Quarter::QuarterWidget::viewportEvent(QEvent*) () from 
/usr/lib/freecad-python2/lib/libFreeCADGui.so
#16 0xb3b88090 in 
QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, 
QEvent*) () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#17 0xb434cda0 in QApplicationPrivate::notify_helper(QObject*, QEvent*) 
() from /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
#18 0xb43552a8 in QApplication::notify(QObject*, QEvent*) () from 
/usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
Backtrace stopped: Cannot access memory at address 0x9a512f8
(gdb) q
A debugging session is active.

        Inferior 1 [process 7640] will be killed.

Quit anyway? (y or n) y

--- System information. ---
Architecture:
Kernel: Linux 4.19.50-v7l+

Debian Release: 10.0
500 testing raspbian.raspberrypi.org
500 testing archive.raspberrypi.org

--- Package information. ---
Depends (Version) | Installed
==============================-+-===========
freecad-python2 | 0.18~pre1+dfsg1-5+rpi1
OR freecad-python3 |

Recommends (Version) | Installed
===========================-+-===========
calculix-ccx | 2.11-1
graphviz | 2.40.1-6

Suggests (Version) | Installed
=============================-+-===========
freecad-doc |
povray |
python-collada |

XECDesign, 2019-08-13:
Being tracked upstream - https://freecadweb.org/tracker/view.php?id=4083

XECDesign, 2022-01-18:
All the solutions I see are either "Use Raspberry Pi OS 64bit" or "Build against Qt4", neither of which are a fix we can use going forward. Some say "Raspbian needs to update their libs" without going into detail or what libs needs to be updated and how. A link claiming to contain a fix links to a forum post which links back to the issue containing the original comment.

If nobody upstream has any idea what the issue is, I certainly don't have a chance.

I've left a comment on the linked issue, hoping for some follow up, but if no suitable solution comes up, this is a wontfix.