RPi-Distro/repo #144: Sendmail authentication error on Stretch

BokeMoire, 2019-07-14:
On my existing Wheezy, aptitude shows that libsasl2-modules was automatically installed:
i A libsasl2-modules                        - Cyrus SASL - pluggable authentication modules

This was not installed automatically on Stretch:
i   libsasl2-modules                            - Cyrus SASL - pluggable authentication modules

I was having authentication problems when attempting to use sendmail on the PI as an outgoing server from Thunderbird on my mac, mail on my android phone and alpine using ssh to the PI.  

Installing libsasl2-modules solved the problem.

XECDesign, 2019-07-15:
sendmail does not come from our repo. It's an unmodified package from debian.

libsasl2-modules is listed as a suggested, but not required or recommended dependency of sendmail. So it doesn't sound like a bug, just a decision by the maintainers of the package.

BokeMoire, 2019-07-15:
XECDesign wrote to RPi-Distro/repo
[at 09:51 (-0000) on Monday, 15th July, 2019]:

> sendmail does not come from our repo. It's an unmodified package from debian.
>
> libsasl2-modules is listed as a suggested, but not required or 
> recommended dependency of sendmail. So it doesn't sound like a bug, just 
> a decision by the maintainers of the package.
>
>

The problem is that I missed it because it DOESN'T appear to be suggested.

Package: sendmail
Version: 8.15.2-12
State: installed
Automatically installed: no
Priority: optional
Section: mail
Maintainer: Debian QA Group <packages@qa.debian.org>
Architecture: all
Uncompressed Size: 240 k
Depends: sendmail-base, sendmail-bin, sendmail-cf, sensible-mda
Suggests: sendmail-doc, rmail
Description: powerful, efficient, and scalable Mail Transport Agent 
(metapackage)

Package: sasl2-bin
Version: 2.1.27+dfsg-1+b1
State: installed
Automatically installed: no
Priority: optional
Section: utils
Maintainer: Debian Cyrus Team <team+cyrus@tracker.debian.org>
Architecture: armhf
Uncompressed Size: 374 k
Depends: db-util, libsasl2-2 (>= 2.1.27+dfsg-1+b1), lsb-base (>= 3.0-6), 
debconf (>= 0.5) |
          debconf-2.0, libc6 (>= 2.28), libdb5.3, libkrb5-3 (>= 
1.13~alpha1+dfsg), libldap-2.4-2 (>=
          2.4.7), libpam0g (>= 0.99.7.1), libssl1.1 (>= 1.1.0)
PreDepends: init-system-helpers (>= 1.54~)

XECDesign, 2019-07-15:
suggested by sendmail-bin, which is a dependency of sendmail

BokeMoire, 2019-07-15:
XECDesign wrote to RPi-Distro/repo
[at 13:44 (-0000) on Monday, 15th July, 2019]:

> suggested by sendmail-bin, which is a dependency of sendmail
>
>

Hmmm, sendmail-bin has 11 depends so I should really now be decending the 
tree into their suggests.  And then their depends ...  suggests ... 
depends ... for ever.

Maybe the libsasl2-modules suggest could be placed earlier in the tree?

XECDesign, 2019-07-16:
That's up to the maintainer of the package and probably too late to change anything like that for Buster.

closed 2019-07-23 by XECDesign