RPi-Distro/repo #145: 100% CPU by pcmanfm

basilrabi, 2019-07-28:
From time to time, the load of 1 cpu core becomes 100% due to `pcmanfm --desktop --profile LXDE-pi` according to `htop`. This is not really reproducible and seems to happen randomly. According to [some cases](https://www.raspberrypi.org/forums/viewtopic.php?t=148376), the 100% cpu load is caused by `matchbox-keyboard` however I'm not using that.

```
More info
hardware: RPi 3B
OS: Raspbian Buster
kernel: 4.19.58-v7+ #1245 SMP
```

spl237, 2019-07-31:
This is going to be hard to isolate from that information; I've never seen it happen here.

The only thing I can think that pcmanfm does that might cause a high CPU load is mounting or unmounting external (USB) drives - if a device which is being used via USB suddenly vanishes, things can get stuck in a high CPU load state. (About the only reliably reproducible method I have found to stick a CPU core at 100% is to pull out a USB audio device when it is in use.)

Are you by any chance using external USB devices? Are you perhaps pulling them out without ejecting them first? I wonder if it is something like that.

basilrabi, 2019-07-31:
I think you're right. I'm using an external hard disk via USB and because of the loose cable/connection, the disk disconnects when the cable is touched. Thank you.

spl237, 2019-07-31:
OK, in that case there probably isn't much we can do to fix it - the USB interface isn't very helpful when it comes to reporting errors. I'll have a look and see if there is anything sensible that pcmanfm can do to try and fail in a more benign fashion.

anordal, 2020-04-05:
I just got this when playing music from sshfs, without any connection problem.
I only disconnected the bluetooth headset temporarily when the phone rang. That's not relevant, is it?

spl237, 2020-04-05:
> I only disconnected the bluetooth headset temporarily when the phone rang. That's not relevant, is it?

It probably is - at a guess, when the headset was disconnected, ALSA kept a handle open to it, as it doesn't have a mechanism for detecting devices no longer being available. An open ALSA handle with nothing for it to talk to seems to cause high CPU loads.

anordal, 2020-04-05:
But why does pcmanfm care about alsa?

Also, I'm using pulseaudio. I can see that my player (mpv) is using pulseadio, but I'm not able to judge by the look of `pactl list` whether it is using alsa as a backend.

spl237, 2020-04-05:
I suspect it isn't pcmanfm in your case which is causing the high CPU load - pcmanfm goes into a high load state if a USB device is unexpectedly disconnected. Audio devices being unexpectedly disconnected causes a similar high CPU load, but usually in lxpanel or one of its plugins.

If you're using pulseaudio, you're on your own! It's not a configuration we support. But yes, it does use ALSA as a backend; pulseaudio is just a front-end for ALSA designed to address some of ALSA's limitations - which it does by introducing a different set of broken things...

anordal, 2020-04-05:
I saw the same `pcmanfm --desktop --profile LXDE-pi` process using 100% CPU in htop.

spl237, 2020-04-05:
As above - in that case, you probably had a glitch which resulted in a drive being disconnected. It happens... Unless you have a reproducible test case, there's not much we can do - there are known issues with device disconnection causing this.

Serondrych, 2020-04-07:
I just had the same issue, in my case pcmanfm was connected to a server via ssh, disconnecting the server did not produce change. Killing pcmanfm process returned cpu load to normal.

sgbeal, 2020-08-24:
FWIW, this is happening to me on a system which boots an RPi4B to Raspberry Pi OS from an external SSD-over-USB disk but which i have not physically touched (operating it only via ssh). A glitch caused by a loose cable can be effectively ruled out: not only is the system not touched, but that's the only filesystem, so a disconnect would likely trigger a system panic.

It's happened twice in the past 24 hours, and i only noticed when my pihole web interface showed unexpected CPU load.

`apt show` says this is version 1.3.1-1+rpt22.

kubycsolutions, 2021-06-09:
One of the other distributions (bodhi?) recently commented that this was known to happen when USB drives were unplugged; apparently ejecting a CD has similar effects.

They also say they just merged a fix for it.

Could whoever owns pcmanfm on raspbian *PLEASE* investigate and merge?

spl237, 2021-06-09:
> One of the other distributions (bodhi?) recently commented that this was known to happen when USB drives were unplugged; apparently ejecting a CD has similar effects.
> 
> They also say they just merged a fix for it.
> 
> Could whoever owns pcmanfm on raspbian _PLEASE_ investigate and merge?

I'm happy to merge a fix if there is one, but the information you have provided above is not going to be enough for me to do so - can you please find the actual comment and post a link? Just saying "one of the other distributions has a fix" really doesn't narrow things down much...

sgbeal, 2021-09-14:
Just FYI, this is still happening at random intervals - sometimes once a week and sometimes twice a day. The only indication that it's happening on my terminal is that the Argon case's fan kicks in at 50C (which it never does under normal use, as this pi4 is just operating as a remote terminal so has no appreciable load). The only known workaround is to kill the process and manually restart pcmanfm.

```
$ apt show pcmanfm
Package: pcmanfm
Version: 1.3.1-1+rpt27
```

Raspberry Pi OS version is current as of about an hour ago.

kubycsolutions, 2021-09-14:
Alternatively, if anyone has a better way to set up “new CD has been inserted, so auto-launch abcde to rip it using my standard settings” (which is my actual goal) and avoid pcmanfm entirely, that would be good too.

()  Plaintext Ribbon Campaign
/\  Stamp out HTML e-mail!

bggardner, 2021-09-20:
@spl237  I'm not sure if @kubycsolutions  meant this bug/fix when referring to Bodhi https://github.com/lxde/menu-cache/issues/7, but it doesn't fix the problem, as I'm running lib-menu-cache 1.1.0 and the issue still exists.  However, it does seem to be very similar, as I'm thinking it has more to do with the "Open With" dialog than USB (which spawns when a USB drive is automounted).

sgbeal, 2021-09-21:
FWIW, it happens to me without me ever using any USB drives other than the one i boot from. Nor have i ever used the "open with" file manager feature. It just random happens from time to time even when no significant local activity is going on (this device is mostly a terminal for remote X11 clients). i don't recall it ever triggering while i was actively using the file manager.

reticulatingsplines, 2021-11-03:
Adding that this is happening to me as well after moving to an external USB3.0 hard drive (with its own power supply, just as a note). No obvious cause, one core is just burning up at 100% thanks to pcmanfm and only connection/open display is a remote terminal session via ssh.

`$ apt show pcmanfm
`Package: pcmanfm
`Version: 1.3.2-1+rpt5
`
`$ cat /etc/os-release
`PRETTY_NAME="Debian GNU/Linux bookworm/sid"
`NAME="Debian GNU/Linux"
`ID=debian
`
`$ cat /etc/rpi-issue
`Raspberry Pi reference 2021-05-07

ghollingworth, 2021-11-03:
Can you confirm the make/model/size of the hard drive and we'll try and reproduce here...

Gordon

sgbeal, 2021-11-03:
FWIW, `dmesg` on my system says it's a `SanDisk SSD PLUS 480GB`.

It's still highly unpredictable, seemingly impossible to reproduce on demand.

kubycsolutions, 2021-11-03:
On my system, where I was trying to use it to trigger a shell script when the CD was inserted, that action kicked it into this spin loop pretty reliably. Not every time, but much more often than not. Might or might not do the same on other systems or with other USB CD drives; your mileage, millage, and melange may vary.

reticulatingsplines, 2021-11-03:
I have a Seagate "Mobile" 2.5" 1TB HDD (5400rpm, 128mb cache, part no. 1RK172-568) inside of a Sabrent USB 3.0 SATA enclosure (part no. EC-DFFN, separate DC power supply) connected to one of the two USB 3.0 ports on the pi.

Geotv123, 2022-01-18:
Same issues here. Rpi 4b, latest Raspberry Pi Os booting from USB3 ssd in a 2.5’ Vantech USB3 and esata enclosure. I also plug an iPhone via lightning cable once in a while and it doesn’t show up eject or Disks etc, yet Removable medium is inserted box pops up with Open in File Manager option.

XECDesign, 2022-02-17:
I think unless we have steps to reliably reproduce the issue, there's not much we can do. Closing for now.

SodaWithoutSparkles, 2022-04-17:
I also have the same issue

Pi4 2GB with latest software

After reading this, I believe plugging in a phone via USB and deny MTA access would cause the issue. I am trying to do ADB while doing that.

automaciej, 2022-05-01:
I can reproduce this reliably on Raspberry PI. To reproduce:

1. Connect a CD drive via USB
2. Wait for a dialog "what to do with this CD?" to appear, click Cancel
3. Start VLC, press CTRL+D, select audio CD, play a few seconds
4. Exit VLC
5. Open the pcmanfm window, click unmount on one of the two Audio CD icons (a single CD appears as two)
6. Now one of the CPUs is a 100%
7. On screen, there's this message printed repeatedly as fast as the computer can manage: `** (pcmanfm:61759): WARNING **: 14:16:40.742: got unmount for unknown desktop item`

I ran it under gdb, and printed a stack trace.

```
** (pcmanfm:61759): WARNING **: 14:16:40.742: got unmount for unknown desktop item

** (pcmanfm:61759): WARNING **: 14:16:40.743:^C
Thread 1 "pcmanfm" received signal SIGINT, Interrupt.
__GI___libc_write (nbytes=104, buf=0x7fffffc358, fd=<optimized out>) at ../sysdeps/unix/sysv/linux/write.c:26
26	../sysdeps/unix/sysv/linux/write.c: No such file or directory.
(gdb) where
#0  __GI___libc_write (nbytes=104, buf=0x7fffffc358, fd=<optimized out>) at ../sysdeps/unix/sysv/linux/write.c:26
#1  __GI___libc_write (fd=<optimized out>, buf=buf@entry=0x7fffffc358, nbytes=nbytes@entry=104) at ../sysdeps/unix/sysv/linux/write.c:24
#2  0x0000007ff6d86394 in _IO_new_file_write (f=0x7ff6e844a0 <_IO_2_1_stderr_>, data=0x7fffffc358, n=104) at fileops.c:1181
#3  0x0000007ff6d8573c in new_do_write
    (fp=fp@entry=0x7ff6e844a0 <_IO_2_1_stderr_>, data=data@entry=0x7fffffc358 "\n** (pcmanfm:61759): \033[1;33mWARNING\033[0m **: \033[34m14:16:40.748\033[0m: got unmount for unknown desktop item\n\340\303\377\377\177", to_do=to_do@entry=104) at libioP.h:948
#4  0x0000007ff6d86aac in _IO_new_file_xsputn (n=104, data=<optimized out>, f=0x7ff6e844a0 <_IO_2_1_stderr_>) at fileops.c:1255
#5  _IO_new_file_xsputn (f=0x7ff6e844a0 <_IO_2_1_stderr_>, data=<optimized out>, n=104) at fileops.c:1197
#6  0x0000007ff6d72ac0 in buffered_vfprintfPython Exception <class 'gdb.MemoryError'> Cannot access memory at address 0x68: 
 (s=s@entry=0x7ff6e844a0 <_IO_2_1_stderr_>, format=format@entry=0x7ff6f714d8 "%s\n", args=#7  0x0000007ff6d7059c in __vfprintf_internal
    (s=0x7ff6e844a0 <_IO_2_1_stderr_>, format=0x7ff6f714d8 "%s\n", ap=..., mode_flags=713708032) at vfprintf-internal.c:1346
#8  0x0000007ff6df6d00 in ___fprintf_chk (fp=<optimized out>, flag=<optimized out>, format=<optimized out>) at fprintf_chk.c:33
#9  0x0000007ff6f16628 in g_log_writer_standard_streams () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#10 0x0000007ff6f16754 in g_log_writer_default () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#11 0x0000007ff6f14ab8 in g_log_structured_array () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#12 0x0000007ff6f14f34 in g_log_default_handler () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#13 0x0000007ff6f151b0 in g_logv () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#14 0x0000007ff6f15440 in g_log () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#15 0x000000555557983c in on_idle_extra_item_remove (user_data=0x7fe8022a00) at desktop.c:976
#16 0x0000007ff750fad8 in  () at /lib/aarch64-linux-gnu/libgdk-3.so.0
#17 0x0000007ff6f0dab4 in g_main_context_dispatch () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#18 0x0000007ff6f0de5c in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#19 0x0000007ff6f0e1b0 in g_main_loop_run () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#20 0x0000007ff7862b50 in gtk_main () at /lib/aarch64-linux-gnu/libgtk-3.so.0
#21 0x0000005555565838 in main (argc=<optimized out>, argv=<optimized out>) at pcmanfm.c:286
```

ghollingworth, 2022-05-02:
Looks likely that this is GVolumeMonitor in gio causing the problem...

https://github.com/lxde/pcmanfm/blob/2c0a6b80dab1cd97f0a408ebbfa2f3e738f80648/src/desktop.c#L720

Is being triggered...  This is happening because there is a callback being called on a signal for mount-removed (which is triggered by GVolumeMount) as set up here:

https://github.com/lxde/pcmanfm/blob/2c0a6b80dab1cd97f0a408ebbfa2f3e738f80648/src/desktop.c#L5911

spl237, 2022-05-11:
I'm afraid I have just tried the procedure described above using a USB CD drive and an audio CD, and when I do exactly the same steps, I get no unusual CPU load, and no error message from pcmanfm; unmounting the audio disc just means it vanishes from pcmanfm with no other symptoms seen. I've repeated the test several times; no sign of the abnormal CPU load here. I also do not get any of those messages or any abnormal CPU load if I just disconnect the drive while a disc is playing, or disconnect any other USB drives without ejecting, come to that. It may be something specific to the CD-ROM drive or USB interface being used.

I'll investigate the GVolumeMonitor calls to see if I can see anything odd going on there.

spl237, 2022-05-13:
In order to help diagnose this, it would be very helpful if anyone who regularly sees this problem could do the following:

1. Download the modified pcmanfm deb file from https://drive.google.com/file/d/1L2NcanmnFB0I44Y27_O0xMhtuxTz48X0/view?usp=sharing This has had some extra logging code added to it which keeps track of mount and unmount events; it is otherwise identical to the standard version.
2. Install it by copying it to your Pi and doing "sudo dpkg -i pcmanfm_1.3.2-1+rpt8_armhf.deb" in the directory where you have put the file. Then reboot.
3. If your Pi then goes into a high CPU load state, please open the file /tmp/pcman.log and post the contents here. Please also provide details of your setup - what version of Pi you are using, what USB peripherals you have connected, what power supply you are using, and any other details about your hardware which may be relevant.

Many thanks in advance. (BTW, the deb provided above is only for 32-bit systems and should not be installed on a Pi running the 64-bit version of the OS - if you are using 64-bit and see this problem, please post back here, and I will create a 64-bit version as well.)

Geotv123, 2022-05-14:
Thank you, I downloaded and installed the deb file. Rebooted. Plugged in my iPhone 7plus via the Usb2.0 ports, unplugged it and it was ok. Plugged it in again and unplugged it. CPU usage around 30% in Task Manager and pcmanfm at the top. Htop shows the last one (3) of 4 (0-3) cpu cores at 100%, rest are fine.

I’m running a pi 4 b, with 4g of Ram. 1T SSD through a Sata enclosure identified as JMicron JMS56x Series. (Vantech 2.5” Usb3 E-sata model). Logitech K400 keyboard with USB receiver in Usb2.0 port. Apple lightning cable in the other Usb2.0 port.

here is the log plus lsusb command:

mount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:29:10 2022
mount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:29:10 2022
mount	/run/user/1000/gvfs/gphoto2:host=Apple_Inc._iPhone_70968e7ddb25ea910df9270ba8f6f413dc62af6d	no_drive_id	no_drive	Fri May 13 20:29:10 2022
mount	/run/user/1000/gvfs/gphoto2:host=Apple_Inc._iPhone_70968e7ddb25ea910df9270ba8f6f413dc62af6d	no_drive_id	no_drive	Fri May 13 20:29:11 2022
unmount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:29:20 2022
unmount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:29:20 2022
unmount	/run/user/1000/gvfs/gphoto2:host=Apple_Inc._iPhone_70968e7ddb25ea910df9270ba8f6f413dc62af6d	no_drive_id	no_drive	Fri May 13 20:29:20 2022
unmount	/run/user/1000/gvfs/gphoto2:host=Apple_Inc._iPhone_70968e7ddb25ea910df9270ba8f6f413dc62af6d	no_drive_id	no_drive	Fri May 13 20:29:20 2022
mount	/run/user/1000/gvfs/gphoto2:host=Apple_Inc._iPhone_70968e7ddb25ea910df9270ba8f6f413dc62af6d	no_drive_id	no_drive	Fri May 13 20:29:39 2022
mount	/run/user/1000/gvfs/gphoto2:host=Apple_Inc._iPhone_70968e7ddb25ea910df9270ba8f6f413dc62af6d	no_drive_id	no_drive	Fri May 13 20:29:39 2022
mount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:29:39 2022
mount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:29:40 2022
unmount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:30:10 2022
unmount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:30:10 2022
mount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:30:11 2022
mount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:30:11 2022
unmount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:30:12 2022
unmount	/run/user/1000/gvfs/afc:host=70968e7ddb25ea910df9270ba8f6f413dc62af6d,port=3	no_drive_id	no_drive	Fri May 13 20:30:12 2022
unmount	/run/user/1000/gvfs/gphoto2:host=Apple_Inc._iPhone_70968e7ddb25ea910df9270ba8f6f413dc62af6d	no_drive_id	no_drive	Fri May 13 20:30:14 2022
unmount	/run/user/1000/gvfs/gphoto2:host=Apple_Inc._iPhone_70968e7ddb25ea910df9270ba8f6f413dc62af6d	no_drive_id	no_drive	Fri May 13 20:30:14 2022

>lsusb

Bus 002 Device 002: ID 152d:0565 JMicron Technology Corp. / JMicron USA Technology Corp. JMS56x Series
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Geotv123, 2022-05-14:
Btw, should I leave or uninstall the deb provided. Not sure how to go about reverting back or if it’s necessary.

Geotv123, 2022-05-14:
Another thing I noticed is that the iPhone gets a new Device number Id every time I plug and unplug. I verified through ‘lsusb’ command. Documents and IPhone(camera icon), in file manager side panel, appear twice or three times if I plug and unplug the device. Ejecting some of them gives out an error sometimes. I hope it helps

spl237, 2022-05-14:
Many thanks for that testing - we'll review the log you provided and see if it gives us any pointers. 

One further question - how long did you leave it between the 100% CPU state and copying the log? I'd just like to get an idea of for how much of the log file it is showing the 100% state.

> Btw, should I leave or uninstall the deb provided. Not sure how to go about reverting back or if it’s necessary.

It's probably best to revert back to the standard version - in a terminal, do "sudo apt install pcmanfm=1.3.2-1+rpt7" - that should just reinstall the version in apt.

Many thanks again for the testing.

Geotv123, 2022-05-14:
> Many thanks for that testing - we'll review the log you provided and see if it gives us any pointers.
>
> One further question - how long did you leave it between the 100% CPU state and copying the [...]

I grabbed the log a minute or two after the cpu load jump.
I could do it again and keep it longer if it helps.

spl237, 2022-05-14:
> I grabbed the log a minute or two after the cpu load jump. I could do it again and keep it longer if it helps.

No, that's fine - no need to do any more testing. I have now been able to repeat the state here by connecting my iPhone. The CPU drain seems to being caused by some spurious logging messages which are being generated in spite of the error to which they pertain not actually happening as far as I can see! That's been really helpful - thank you very much.

spl237, 2022-05-23:
A fix for this has now been pushed to apt - if you update pcmanfm to the latest version, you should find that the 100% CPU has now stopped.

There are still probably some problems in the underlying file system and volume management, which will be investigated as time permits, but the high-level issue should now be fixed, so closing this for now. If anyone sees the high CPU usage with the latest version, please re-open with details.

closed 2022-05-23 by spl237

kubycsolutions, 2022-10-11:
> Just an update; I've found what I think is the root cause of the drain; it's actually a side effect of a fix I put in for a different problem with the file manager, whereby it was possible for drives to remain shown on the desktop after the device they were on had been removed. I'm testing possible fixes at the moment, and will hopefully have an update in the next few days.

Excellent, Simon! Many thanks for your work on this. I'd rather minimize the number of Linux builds I'm working with, for obvious reasons, and having this fixed would let me get back to setting up that machine with a consistent environment.--Sent from phone; apologies for typos and auto incorrections.   /_  Joe Kesselman (he/him/his)-/ _) My Alexa skill for New Music/New Sounds fans:   /   https://www.amazon.com/dp/B09WJ3H657/() Plaintext Ribbon Campaign/\ Stamp out HTML mail!

SodaWithoutSparkles, 2023-06-05:
Ok, so i started having this issue again. I got alerted by the pi-hole admin GUI on high temps (~70C). So I went on to check the process monitor (btop) over VNC. I am surprized to find that pcmanfm is again taking 100% - 102% of the CPU and causing the overheat.

The problem is, I have NOT plugged any USB devices since boot.

I don't know how to capture the logs. I have to issue a SIGTERM to pcmanfm to normal CPU usage. 

I then launch a terminal to try to restart pcmanfm somehow, but soon found that the terminal was unusable, as if somebody held the key `7` on the keyboard. 

After a reboot, the `7` key issue disappeared, so it definitely wasn't the keyboard got stuck.

How can I capture the logs?

AnthonyRKing, 2023-09-17:
Hey really? This is supposed to be fixed? This issue has been affecting my Pi 3B for years on Buster; daily. I have not seen any improvement, as the 100% CPU thing just appears even when the system is idle. I have a few USB sticks attached, wireless mouse/kb by USB and another by Bluetooth. 'Phone gets plugged in to USB & removed without 'eject'. Doing this is not correlated with the 100% issue's appearance. I rarely even actually touch the Pi physically, USB switching being done via a hub.

Well, I notice the PCManFM version I have is older than that reported by @reticulatingsplines in Nov 2021 (actually 1.3.1-1+rpt27, same as @sgbeal reported as being problematic). Could it be that the fixed version simply hasn't made its way to the (Legacy Support) Buster repository?

```
$ cat /etc/rpi-issue 
Raspberry Pi reference 2021-12-02

$ apt-cache policy pcmanfm
pcmanfm:
  Installed: 1.3.1-1+rpt27
  Candidate: 1.3.1-1+rpt27
  Version table:
 *** 1.3.1-1+rpt27 500
        500 http://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status
     1.3.1-1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
```

mrwicks, 2024-02-16:
I ran into this problem.  Just killed pcmanfm with sudo kill -9 pcmanfm.  This just restarts the window manager and doesn't kill your session.

I would debug this and fix it, if there was a reasonable chance the modification was included.  It seems to happen a lot on my pi, but I'm also using an older version of raspberry pi OS so this issue might be fixed.  Should be a simple matter to use Valgrind or gprof to find out where it's spinning.

AnthonyRKing, 2024-02-16:
My understanding is that it has been fixed from Bullseye on, but that support doesn't extend to fixing it for Buster and earlier.