RPi-Distro/repo #148: vlc regression: missing support for MMAL video output on Raspberry

morfeus2, 2019-08-28:
vlc feature regression on Raspbian stretch.
Upgrading from 3.0.7-0+rpt1+deb9u1 to 3.0.8-0+deb9u1 the video output module named "MMAL x11 splitter fro Raspberry Pi" disappeared.
Unfortunately without it, vlc on Raspberry is totally useless.
Maybe it's just a ./configure switch missing.

popcornmix, 2019-08-28:
Are you up to date? See: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=246575&start=25#p1526717

morfeus2, 2019-08-28:
Sorry, I just followed the suggestion received here:
https://bugs.launchpad.net/raspbian/+bug/1841417
without noticing the updates. (when I first submitted I could not find any
comment related to the bug.
+rpt1 is working properly: thanks for fixing it!

closed 2019-08-28 by XECDesign