RPi-Distro/repo #149: raspberrypi-kernel should include kernel version in package metadata

replicajune, 2019-09-08:
The kernel package on most linux systems include the version of the provided kernel. Currently on ubuntu it shows :

``` text
→  dpkg -s linux-generic
...
Version: 5.0.0.27.28
```

On raspbian, the version only seems to be the revision number of the build an the package revision number :

``` text
→  dpkg -s raspberrypi-kernel
...
Version: 1.20190819-1
...
```

No relation can be made to the current running kernel with `uname -r` and the kernel package provided by raspbian as is.

---

The package should have to some extends the kernel information reported by `uname -r`. Currently on raspbian, uname shows :

```
→  uname -r
4.19.57-v7+
```

Then the package could have the version number *4.19.57-1.20190819-1*

XECDesign, 2019-09-09:
Would you settle for the kernel version being included in the package description rather than the package version?

replicajune, 2019-09-09:
The description would hardly being parsable, it's also not so standard. There's no way to build upon the infos available in description to do anything else that just print the info related to a package

To give a bit more info about how i came up to that issue:
When I tried to check what was the running kernel and the last available installed kernel package, I was not able to find the kernel version information using dpkg (which is what I do on ubuntu) the principle is also similar in fedora.

As a workaround right now i'm doing this :
``` sh
      kernel_pkg_ver="$(
        for VER in /lib/modules/*; do
          if [ "${kernel_live_ver}" = "${VER}" ]; then
            echo "${VER}"
            break
          fi
        done
```

`kernel_live_ver` being a result of `uname -r`. the logic is that if I don't have anything ressembling the current running kernel version, I had an update. I rely on the fact that I'll always have either current kernel version number in there or more recent but never older (which would be workable but it's another level of sketchiness to add .. :D )

The end goal is to have something to help automate updates and reboot only when needed, hence the need to have a stable parsable version somewhere.

Are we 100% sure that the version in /lib/modules could also be a valid kernel version number ?
I bet there is an issue changing the scheme of the version number of a package so I understand it's might not be a possible solution.

Can we update a file somewhere with the always last available kernel version ? how to standardize that if so ?

XECDesign, 2019-09-09:
Why wouldn't the description be parsable?

I could put the kernel version number in the package changelog.

Or, if you can rely on having internet access, you can get it from here (with the appropriate version number in the URL) : https://raw.githubusercontent.com/raspberrypi/firmware/1.20190819/extra/uname_string7l

replicajune, 2019-09-09:
It's not the goal of the description field to have parsable items. and it could eventualy be changed.

the file uname_string* looks promising but I can't rely on internet to check if I need to reboot a machine. if I don't have the newer kernel for instance, I don't need to reboot. And for people having tons of raspberry-pies (what's the plural ? :D) it woudln't make sense to update all of them at the same time. So you can't rely on something external to the machine you're on.

That said, if this file is accessible on the machine, it'll do :) then i'll be sure that it's from the package, not from the kernel currently running (ie: not in /proc) and I won't need to rely on something external too.

XECDesign, 2019-10-01:
Kernel version added to the changelog:

```
$ zcat /usr/share/doc/raspberrypi-kernel/changelog.Debian.gz | head

raspberrypi-firmware (1.20190925-1) buster; urgency=medium

  * firmware as of 01508e81ec1e918448227ca864616d56c430b46d
  * Linux version 4.19.75
```

GeoffMaciolek, 2019-10-05:
Thanks for adding this to the changelog!

I do want to second @replicajune on this, though - requiring scripts to parse what's supposed to be a human-readable descriptive field is a bad idea. That's the kind of thing that leads to really fragile automation scripts, unexpected data ending up in places, etc.

I do realize that changing the version numbering scheme could potentially be a breaking change, but the current versioning scheme is out of line with decades of convention among Linux distributions & systems administration.

I tend to vote for the "break things once" option, rather than tbe "leave something kludgy that requires long terrn workarounds." (This is probably die to me having inherited a lot of those long-term workarounds over the years.)

XECDesign, 2019-10-05:
And this is the package I've inherited. As much as I'd love to redo it entirely, it's just not going to be at the top of my todo list in the near future.

closed 2019-10-05 by XECDesign