RPi-Distro/repo #154: update dhcpcd package to fix IPv6 LL race condition

maxnet, 2019-10-23:
Some folks reported not receiving an IPv6 link local address when using usbbootgui/gpioexpander in combination with a Pi 3B+: https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=254541

This turned out to be a timing specific race condition in dhcpcd.
Reported it to author, and he fixed it here: https://roy.marples.name/cgit/dhcpcd.git/commit/?id=b2b6541fe9c067de1f21f0669c4e47cf3c163024

So it would be nice if dhcpcd package was updated to latest version + this patch.

XECDesign, 2019-10-23:
Which version are you referring to as the latest stable? Do we need to jump all the way to 8.1.1?

maxnet, 2019-10-23:
Will leave that up to you.
Patch applies to 7.x branch as well.
But keep in mind that dhcpcd supports a zillion different network configurations, and it's hard to test, if there aren't any side effects when combining old and new code.

XECDesign, 2019-10-23:
Alright, I'll put 8.1.1 with that patch on an internal apt repo for testing for a bit.

XECDesign, 2019-10-23:
So far I've seen it segfault once but haven't managed to get it to happen again. Made available for wider testing here: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=255072

XECDesign, 2019-10-30:
No issues reported or found while testing, but right after I pushed it to the public repo, I saw this:
https://roy.marples.name/archives/dhcpcd-discuss/0002668.html

Leaving it at 8.1.1 for now and keeping an eye on the mailing list, unless we also start getting reports of 100% CPU usage.

XECDesign, 2019-11-05:
Had to roll back due to a large number of people experiencing the 100% CPU use. I'll wait until that and the segfaults are fixed before making another attempt.

maxnet, 2019-11-13:
Supposedly fixed in dhcpcd-8.1.2

XECDesign, 2019-11-18:
Should be fixed now

closed 2019-12-13 by XECDesign