RPi-Distro/repo #155: Can't install libasound2-dev

bhawkins, 2019-11-04:
I'm trying to install libasound2-dev on buster on my rpi2. My guess is that the problem is that RPi customized the binary packages but not the header package, so the dependencies can't be satisfied unless I ditch the -rpt1 packages. Is there a workaround?

```
$ sudo apt-get install -f                                                      
Reading package lists... Done                                                  
Building dependency tree                                                       
Reading state information... Done                                              
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.                                                                      
$ sudo apt-get install libasound2-dev                                          
Reading package lists... Done                                                  
Building dependency tree                                                       
Reading state information... Done                                              
Some packages could not be installed. This may mean that you have              
requested an impossible situation or if you are using the unstable             
distribution that some required packages have not yet been created             
or been moved out of Incoming.                                                 
The following information may help to resolve the situation:                   
                                                                               
The following packages have unmet dependencies:                                
 libasound2-dev : Depends: libasound2 (= 1.1.8-1)                              
E: Unable to correct problems, you have held broken packages.                                                              
$ sudo dpkg --get-selections | grep hold                                                                                            
$ sudo aptitude install libasound2-dev                                         
The following NEW packages will be installed:                                  
  libasound2-dev{b}                                                            
0 packages upgraded, 1 newly installed, 0 to remove and 0 not upgraded.        
Need to get 145 kB of archives. After unpacking 546 kB will be used.           
The following packages have unmet dependencies:                                
 libasound2-dev : Depends: libasound2 (= 1.1.8-1) but 1.1.8-1+rpt1 is installed
The following actions will resolve these dependencies:                         
                                                                               
     Keep the following packages at their current version:                     
1)     libasound2-dev [Not Installed]                                          
                                                                               
                                                                               
                                                                               
Accept this solution? [Y/n/q/?] q                                              
Abandoning all efforts to resolve these dependencies.                          
Abort.                                                                         
```

XECDesign, 2019-11-04:
Everything looks okay from here.

Have you run `sudo apt update`?

If so, what's the output of `apt policy libasound2{,-dev}`?

bhawkins, 2019-11-04:
Thanks for following up. Here's the output of apt policy:
```
$ sudo apt update
Get:1 http://mirrors.ocf.berkeley.edu/raspbian/raspbian buster InRelease [15.0 kB]
Fetched 15.0 kB in 1s (15.3 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
$ apt policy libasound2
libasound2:
  Installed: 1.1.8-1+rpt1
  Candidate: 1.1.8-1+rpt1
  Version table:
 *** 1.1.8-1+rpt1 100
        100 /var/lib/dpkg/status
     1.1.8-1 500
        500 http://mirrors.ocf.berkeley.edu/raspbian/raspbian buster/main armhf Packages
$ apt policy libasound2-dev
libasound2-dev:
  Installed: (none)
  Candidate: 1.1.8-1
  Version table:
     1.1.8-1 500
        500 http://mirrors.ocf.berkeley.edu/raspbian/raspbian buster/main armhf Packages
$
```

XECDesign, 2019-11-04:
you seem to be missing the raspberrypi.org repo (in /etc/apt/sources.list.d/raspi.list by default).

bhawkins, 2019-11-05:
I must've messed it up while switching around mirrors. After adding archive.raspberrypi.org I'm able to install libasound2-dev. Thanks!

closed 2019-11-05 by bhawkins