RPi-Distro/repo #157: Request for enhancement: 64-bit kernel headers

andrum99, 2019-11-23:
The 'raspberrypi-kernel' package now includes a 64-bit version of the kernel. It would be useful to have the source package 'raspberrypi-firmware' and the package 'raspberrypi-kernel-headers' updated to include headers for that 64-bit kernel. This will make building out-of-tree modules easier.

XECDesign, 2019-11-25:
It's a bit of a pain.

https://github.com/raspberrypi/linux/issues/3279

andrum99, 2019-11-25:
I hadn't realised this was taken out for a reason. Closing.

closed 2019-11-25 by andrum99

XECDesign, 2019-11-25:
It's on the todo list, but not a top priority.

alkisg, 2020-02-03:
Hi guys, is it possible to reopen this issue until it's fixed?
I just tried to file a new similar one, and fortunately github showed me this as a possible duplicate.

Also, https://github.com/raspberrypi/linux/issues/3279 is closed now, yet the aarch64 headers still don't exist in raspberrypi-kernel-headers 1.20200114-1...

Thank you!

XECDesign, 2020-02-03:
I think it's so far from being a top priority that it's a wontfix. May I ask why you need them? How do you run an aarch64 toolchain in Raspbian?

andrum99, 2020-02-03:
@alkisg https://github.com/raspberrypi/linux/issues/3279 was fixed by removing the 64-bit kernel headers from the 'raspberrypi-kernel-headers' package - the underlying issue still exists.

The workaround to not having 64-bit kernel headers in the  'raspberrypi-kernel-headers' package is simply to download the relevant kernel source from https://github.com/raspberrypi/linux/releases instead.

alkisg, 2020-02-03:
Thank you @XECDesign and @andrum99,

I was asking on behalf of a company that sells wifi dongles that need an out-of-tree module; their Raspbian users sometimes boot with kernel8.img and then they can't compile the module. I'm afraid that fetching kernel headers from https://github.com/raspberrypi/linux/releases would be too much trouble for the users, so I guess it's a no-go for now.

andrum99, 2020-02-03:
There's nothing stopping them packaging the source or headers up themselves and supplying them to their users if they really need it. Another option would be to offer the modules as pre-compiled binaries packages to require specific kernel versions

alkisg, 2020-02-03:
The module installer apt installs the headers; it needs to run in so many different distributions, flavors and kernel versions, that little things like special casing "raspberry-pi-kernels" for Raspbian is acceptable, but including the headers for thousands of kernels isn't really appropriate.

The reasons why users boot with the 64bit kernel vary a lot... sometimes they're just experimenting and think that `rpi-update` (which also lacks headers) or 64bit might bring them better performance, some other times they do need some 64bit only apps etc.

Thank you for your time! :)

XECDesign, 2020-02-04:
Having the headers .deb wouldn't help in that situation since there's no aarch64 toolchain in Raspbian. They won't be able to build the modules that way.

andrum99, 2020-02-04:
> Having the headers .deb wouldn't help in that situation since there's no aarch64 toolchain in Raspbian. They won't be able to build the modules that way.

Good point. I was kind of forgetting I have to use sakaki's systemd-nspawn container for that.

alkisg, 2020-02-04:
Hi guys, thank you for the additional information.

For reference, I also tried with https://images.offensive-security.com/arm-images/kali-linux-2020.1-rpi3-nexmon.img.xz that some of the customers use, and there the aarch64 headers and toolchain were preinstalled and the module compilation was successful without requiring any additional packages at all.

XECDesign, 2020-02-04:
That image seems to be entirely 64bit so it makes sense there.