RPi-Distro/repo #158: IPython (for Python 3) seems to be broken

bernardosulzbach, 2019-12-09:
After installing python3 (3.5.3) and python3-ipython from the repositories and trying to run it, the following issue happens:

```
pi@raspberrypi:~ $ ipython3
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/IPython/__init__.py", line 49, in <module>
    from .terminal.embed import embed
  File "/usr/lib/python3/dist-packages/IPython/terminal/embed.py", line 17, in <module>
    from IPython.terminal.interactiveshell import TerminalInteractiveShell
  File "/usr/lib/python3/dist-packages/IPython/terminal/interactiveshell.py", line 19, in <module>
    from prompt_toolkit.shortcuts import create_prompt_application, create_eventloop, create_prompt_layout, create_output
ImportError: cannot import name 'create_prompt_application'
```

I think this is probably because prompt-toolkit is too outdated in the repositories.

XECDesign, 2019-12-10:
Not sure what's going on with your install, but everything looks okay from here.

The current version of python3 in Buster is 3.7.3. ipython3 installs and runs perfectly fine, even with python3.5 installed on the side.

If you're using pip to override what's in apt, such problems are almost guaranteed.

XECDesign, 2019-12-10:
Just out of curiosity, I tried running ipython through python 3.5.4 and that still works.

bernardosulzbach, 2019-12-10:
Thanks for looking into it @XECDesign. I'm probably doing something wrong then.

Nothing was fetched using pip, everything is from the repositories. Python is version 3.5.3 after updating & upgrading.

I seem to be running Debian 9. Still, it would be expected to work, right?

XECDesign, 2019-12-10:
Although I'd expect it to work under Stretch, I wouldn't be surprised if it didn't. We don't really support older releases, so it's not something I can easily look into.

I can't think of any changed in the archive.raspberrypi.org repo that would break it, so it may be worth checking if it works in vanilla Debian. If it doesn't, maybe it could be worth while reporting the bug to them.

Sorry I couldn't be of much help here.

bernardosulzbach, 2019-12-10:
Yes, if the Stretch repositories are more or less abandoned, things that worked should just keep working.

Guess I'll upgrade to Raspbian Buster then. Thanks for your comments.

closed 2019-12-13 by XECDesign