RPi-Distro/repo #159: linux-perf-4.19 is not available in buster

superfrink, 2019-12-18:
Currently, the linux-perf package is version 4.18 while the kernel is version 4.19.  As a result, perf does not run.

```
[root@p ~]# perf
/usr/bin/perf: line 13: exec: perf_4.19: not found
E: linux-perf-4.19 is not installed.

[root@p ~]# apt-cache search linux-perf
linux-perf - Performance analysis tools for Linux (meta-package)
linux-perf-4.18 - Performance analysis tools for Linux 4.18
linux-perf-4.9 - Performance analysis tools for Linux 4.9

[root@p ~]# uname -a
Linux p 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

[root@p ~]# cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi

[root@p ~]# cat /etc/debian_version
10.2
```

superfrink, 2019-12-18:
As a potential workaround, the package found at https://packages.debian.org/buster/armhf/linux-perf-4.19/download seems like it might work.

davegallant, 2020-07-08:
> As a potential workaround, the package found at https://packages.debian.org/buster/armhf/linux-perf-4.19/download seems like it might work.

On a Pi Zero this leads a `Segmentation fault`.

superfrink, 2020-07-08:
> > https://packages.debian.org/buster/armhf/linux-perf-4.19/download seems like it might work.
> 
> On a Pi Zero this leads a `Segmentation fault`.

That is a good reminder that mixing packages from distros may not work.

It turns out Debian armhf packages are built for ARMv7 and Raspbian armhf packages are built for ARMv6.

The Pi 3 can run both ARMv7 and ARMv6 instructions but the Pi Zero can only run ARMv6 instructions.  The Debian packages won't work on the Pi Zero.

bkil, 2020-08-01:
Same problem for the current kernel, `linux-perf-5.4`. A similar workaround still works:
* https://packages.debian.org/buster-backports/linux-perf-5.4

XECDesign, 2022-02-17:
Closing as a duplicate of https://github.com/RPi-Distro/repo/issues/110

closed 2022-02-17 by XECDesign