RPi-Distro/repo #160: APT distribution template for Raspbian/buster is missing.

Ivan61, 2019-12-22:
I have a problem when I execute ` add-apt-repository`.
```
pi@raspberrypi:~ $ sudo add-apt-repository "deb [arch=armhf] https://download.docker.com/linux/raspbian $(lsb_release -cs) stable"
Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)    
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 93, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/buster
```

```
pi@raspberrypi:~ $ ls -l /usr/share/python-apt/templates/*.info 
-rw-r--r-- 1 root root 11599 Mar 11  2019 /usr/share/python-apt/templates/Blankon.info
-rw-r--r-- 1 root root  8156 Mar 11  2019 /usr/share/python-apt/templates/Debian.info
-rw-r--r-- 1 root root  1590 Mar 11  2019 /usr/share/python-apt/templates/gNewSense.info
-rw-r--r-- 1 root root  1169 Mar 11  2019 /usr/share/python-apt/templates/Tanglu.info
-rw-r--r-- 1 root root 85596 Mar 11  2019 /usr/share/python-apt/templates/Ubuntu.info
```

I try to add one.  I'm not sure that the configuration is correct or not.
But it works after I add it, /usr/share/python-apt/templates/Raspbian.info.

```
pi@raspberrypi:~ $ cat /usr/share/python-apt/templates/Raspbian.info
Suite: buster
RepositoryType: deb
BaseURI: http://raspbian.raspberrypi.org/raspbian/ 
MatchURI: http://raspbian.raspberrypi.org/raspbian/

pi@raspberrypi:~ $ cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
# deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi

pi@raspberrypi:~ $ sudo add-apt-repository "deb [arch=armhf] https://download.docker.com/linux/raspbian $(lsb_release -cs) stable"

pi@raspberrypi:~ $ cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
# deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable
# deb-src [arch=armhf] https://download.docker.com/linux/raspbian buster stable
```

I guess that the source package is from here:
https://packages.debian.org/en/jessie/all/python-apt-common/filelist
https://packages.debian.org/en/stretch/all/python-apt-common/filelist
https://packages.debian.org/en/buster/all/python-apt-common/filelist

Does Raspbian have plan to add it for future release?

tcurdt, 2021-02-14:
This issue is from 2019 and still not fixed? Is there a reason for that?

XECDesign, 2021-02-15:
I'm not familiar with `add-apt-repository`, but this looks like it should've been reported here: https://bugs.launchpad.net/raspbian

closed 2021-03-01 by no-response[bot]

MichaIng, 2021-03-02:
What a timing coincidence. The issue has been practically solved, regarding the Docker install instructions, with this commit: https://github.com/docker/docker.github.io/pull/11990

The instructions do not contain `add-apt-repository` anymore, which still does not work on Raspbian, but at least I don't know any relevant case where it is 😄.

The change is already live: https://docs.docker.com/engine/install/debian/#install-using-the-repository

I opened another PR to add instructions for the Docker Raspbian branch: https://github.com/docker/docker.github.io/pull/12331
That allows following the "Install using the repository" method on Raspbian, respectively the ARMv6 models, where the Debian armv7hf binaries would fail. Currently the docs still state that the manual install method does not work on Raspbian, but the installer must be used instead, which is made obsolete and the statement removed with the above PR 🙂.

DiscGolf, 2021-05-16:
solution worked fine for me