RPi-Distro/repo #161: Packages for Pi Face Control and Display in Raspian Buster missing

leprovokateur, 2019-12-25:
Hi,

The needed packages for the Pi Face Control and Display are missing in Raspian Buster, similar to https://github.com/RPi-Distro/repo/issues/3
Please provide them. 

Best regards,
Robert

GeoffMaciolek, 2019-12-25:
Robert - can you be somewhat more specific? Names of packages, etc? (And, is this problem new to the Buster release?)

leprovokateur, 2019-12-25:
Hi,

With 
deb http://archive.raspberrypi.org/debian/ stretch main
in the /etc/apt/sources.list.d/raspi.list I get with a aptitude search piface, with buster I don't:

p   pifacedigital-emulator                                                                                                                                  - Emulator for the PiFace Digital I/O boards.                                                                                                                      
p   pifacertc                                                                                                                                               - Save/restore system clock with the PiFace real-time-clock board                                                                                                  
p   python-pifacecad                                                                                                                                        - API for PiFace Control and Display boards (Python 2)                                                                                                             
p   python-pifacecad-doc                                                                                                                                    - API for PiFace Control and Display boards (common documentation)                                                                                                 
p   python-pifacecommon                                                                                                                                     - Common functions for PiFace boards (Python 2)                                                                                                                    
p   python-pifacecommon-doc                                                                                                                                 - Common functions for PiFace boards (common documentation)                                                                                                        
p   python-pifacedigitalio                                                                                                                                  - API for PiFace Digital I/O boards (Python 2)                                                                                                                     
p   python-pifacedigitalio-doc                                                                                                                              - API for PiFace Digital I/O boards (common documentation)                                                                                                         
p   python-pifacerelayplus                                                                                                                                  - API for PiFace Relay+ boards (Python 2)                                                                                                                          
p   python-pifacerelayplus-doc                                                                                                                              - API for PiFace Relay+ boards (common documentation)                                                                                                              
p   python3-pifacecad                                                                                                                                       - API for PiFace Control and Display boards (Python 3)                                                                                                             
p   python3-pifacecommon                                                                                                                                    - Common functions for PiFace boards (Python 3)                                                                                                                    
p   python3-pifacedigitalio                                                                                                                                 - API for PiFace Digital I/O boards (Python 3)                                                                                                                     
p   python3-pifacerelayplus                                                                                                                                 - API for PiFace Relay+ boards (Python 3)                                                                                                                          

Also some packages related to python and lirc from the raspi repo are only available in Stretch:

p   python-lirc                                                                                                                                             - LIRC client API (Python 2)                                                                                                                                       
p   python-lirc-dbgsym                                                                                                                                      - Debug symbols for python-lirc                                                                                                                                    
[... cut ...]  for Linux Infra-red Remote Control (LIRC) support                                                                                                
p   python3-lirc                                                                                                                                            - LIRC client API (Python 3)                                                                                                                                       
p   python3-lirc-dbgsym  

Robert

XECDesign, 2020-01-06:
piface no longer provide us with these packages

leprovokateur, 2020-01-06:
I made some tests and it seems that python-lirc is the culprit. There is a dependency on python < 3.6. Besides that, there is a newer version available. A build of that on buster would hopefully fix that.

closed 2020-01-07 by XECDesign

leprovokateur, 2020-01-09:
Would you please provide an updated and built against buster's python version of python-lirc?

XECDesign, 2020-01-09:
I believe their recommendation is to install their packages via pip.

I'd volunteer to do the debian packaging if I didn't have enough work to keep me busy as is.