RPi-Distro/repo #163: Set-up wizard save issue

mystyq, 2019-12-31:
I installed Raspbian Full on RPi4, went through the set-up wizard on first-boot (desktop). On the last page of the installer it prompted me to restart.  I ignored the prompt and went to open a terminal to edit my graphic settings (memory splitting) through `raspi-config`.  I issued `sudo reboot` from the terminal window (set-up wizard prompt still open).  On next boot none of the settings I made through the set-up wizard prompt seem to have been saved.  It prompted me to go through the wizard again.

Not sure if this is a bug, perhaps the set-up wizard makes the final changes when it finishes on close

XECDesign, 2020-01-06:
@spl237

XECDesign, 2020-12-08:
I am guessing this happens if you close the window by pressing the 'x' rather than completing pressing skip at the end. Arguably, that's expected behaviour if you do that?

spl237, 2020-12-08:
Unless the wizard is exited by pressing either the "Cancel" button on the first page or the "Restart", "Later" or "OK" buttons on the final page, it will continue to run at startup.

closed 2020-12-08 by spl237