RPi-Distro/repo #164: Building own kernel deb

rpier1, 2020-02-03:
Hi, 

I would like to **modify the kernel** and **create a deb** from it, 
similar to the official [`raspberrypi-kernel_1.20190925+1-1_armhf.deb`](http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190925+1-1_armhf.deb)
 

Which branch or tag of
&nbsp; &nbsp; &nbsp; &nbsp; https://github.com/raspberrypi/linux
do I need for a **stable kernel** ?
Can I just pick any tag from here https://github.com/raspberrypi/linux/tags ?

I only need the kernel for RPi 3B+, so I build with:
```bash
KERNEL=kernel7
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig # then modify .config for my purposes
make -j$(nproc) ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- deb-pkg
```
... I end up with 

```bash
linux-headers-4.19.97-v7+_4.19.97-v7+-1_armhf.deb # 11M
linux-image-4.19.97-v7+_4.19.97-v7+-1_armhf.deb   # 18M  - smaller than offical raspberrypi-kernel which has 68M
linux-libc-dev_4.19.97-v7+-1_armhf.deb            # 975K
```

Also: comparing the files in the official [`raspberrypi-kernel_1.20190925+1-1_armhf.deb`](http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190925+1-1_armhf.deb) with my own, via `dpkg-deb -c *kernel*.deb` shows differences:

My own kernel only has: 
```
/lib/modules/4.19.97-v7+/
```

while the official kernel has:
```
/lib/modules/4.19.75+/
/lib/modules/4.19.75-v7+/
/lib/modules/4.19.75-v7l+/
/lib/modules/4.19.75-v8+/
```

My own kernel has `dtb`s under `/usr/lib/linux-image-4.19.97-v7+`:
```
./usr/lib/linux-image-4.19.97-v7+/*.dtb
./usr/lib/linux-image-4.19.97-v7+/overlays/*.dtbo
```

while the official kernel has this under `/boot/`
```
./boot/*.dtb
./boot/overlays/*.dtbo

./boot/kernel.img
./boot/kernel7.img
./boot/kernel7l.img
./boot/kernel8.img
...
```

* **How is the official kernel deb [`raspberrypi-kernel_1.20190925+1-1_armhf.deb`](http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190925+1-1_armhf.deb) built?**

When installing my own kernel via deb:
- Am I safe to just install my own kernel `linux-image-4.19.97-v7+_4.19.97-v7+-1_armhf.deb`?
- Or do I need to install **other** debs as well, such as linux-libc-dev: `linux-libc-dev_4.19.97-v7+-1_armhf.deb`?

```bash
raspberrypi-bootloader:armhf
raspberrypi-kernel:armhf       # safe to just replace this with own deb?
raspberrypi-net-mods        
raspberrypi-sys-mods:armhf  
raspbian-archive-keyring    
raspi-config                
linux-libc-dev:armhf
#etc.
```

Thanks for tips.

rpier1, 2020-02-03:
>   * How is the official kernel deb [`raspberrypi-kernel_1.20190925+1-1_armhf.deb`](http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20190925+1-1_armhf.deb) built?

I've seen the comment here: https://github.com/RPi-Distro/repo/issues/104#issuecomment-380429486
... referencing https://github.com/RPi-Distro/firmware/blob/debian/debian/update.sh (There I see that some modules are built, but not the kernel itself. From where is that `update.sh` actually called?)

PS:
I've also seen another project which looks really good:
https://github.com/hypriot/rpi-kernel
(Currently I'm leaning towards going with that project's kernel build).

XECDesign, 2020-02-03:
This repo is for reporting issues with packages in the archive.raspberrypi.org repo. For general support, please use https://www.raspberrypi.org/forums/

closed 2020-02-03 by XECDesign