RPi-Distro/repo #165: Chromium - "Can't update Chromium" notification

IPB-DevOps, 2020-02-11:
hi folks,

seems to be the same bug as this one:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=943668

Full Message in chromium: 
"Can't update Chromium. Chromium couldn't update to the latest version, so you're missing out on latest features and seurity fixes"

hardware: Raspberry Pi 3 Model B Rev 1.2
os version: raspbian 10
kernel: 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
package: chromium-browser
version: Chromium 78.0.3904.108 Built on Raspbian , running on Raspbian 10

best regards

XECDesign, 2020-02-11:
We'll need to pull in this fix for the next build:

https://github.com/chromium/chromium/commit/e79d9d0e06b825d2e62b38db03248c0e6ceec7e4.diff

didiatworkz, 2020-02-12:
Can confirm the bug - Please fix it very fast :/

XECDesign, 2020-02-12:
No fix until we have the next version of chromium ready, which won't be any time soon.

If you require a workaround, you can try this: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=264399#p1610138

didiatworkz, 2020-02-12:
Hi @XECDesign,
thanks - that is a good solution for the moment ;)

trentasis, 2020-02-26:
any scheduled date?

UtechtDustin, 2020-03-31:
Any news here ?
We have the same issue on a production system, we also tried a few parameters e.g. the `--check-for-update-interval`, but the issue still exists for us.

nicemachines, 2020-04-14:
It's now April 14, 2020 and this issue continues!  Seems to me to be a virus because even though you un install and re install, keeps coming back saying it can't update.  You would think that after all this time, someone would get rid of this update if it's not a virus that has found it's way into Buster!

nicemachines, 2020-04-14:
Found a fix, use firefox esr!

UtechtDustin, 2020-04-22:
@XECDesign 
Is there a date for the new Release with the fix ?

XECDesign, 2020-04-22:
Not that I'm aware of, no.

UtechtDustin, 2020-04-22:
Thank you for the quick answer, we have the problem that the workaround menitoned above dont work for us.
If we set a high number the "cant update[...]" message is still shown after a few minutes.
Also i tried to set the number to 1, but the message will not be shown instantly as expected.

aa-telali, 2020-05-29:
Is there any way to set this feature to `false`:

```
static constexpr base::Feature kOutdatedBuildDetector = {
      "OutdatedBuildDetector", base::FEATURE_ENABLED_BY_DEFAULT};

  if (!base::FeatureList::IsEnabled(kOutdatedBuildDetector))
    return false;
```

That's current code so should allow us to bypass the check right?

JesseMeiring, 2020-06-11:
I've been pulling my hair out trying to figure this out, and it turns I wasn't doing anything wrong. Geezus. I have tried changing the apt to mark chromium as a hold with _sudo apt-mark hold chromium-browser_ but the notification keeps appearing. I would love a solution to this issue.

KTibow, 2020-07-12:
As stated on the RPi fourms, Chromium has a set build date, and if it gets out of date by too many days then it complains.

aa-telali, 2020-07-21:
It's been 3 months since someone asked so I don't feel too bad.. but apologies @XECDesign, do you have a clearer idea of when the next release is due?

I've been waiting for an update as I need to give an RPi back to an organisation I made a kiosk for - I updated it to the latest release and now have this issue. If the release isn't happening any time soon then I'll try something else (are there any easy instructions on how to do a build from that PR by any chance?).

aa-telali, 2020-09-02:
Is this a modified version of Chromium specific to the RPi or just the Debian version repackaged? The repo looks like it's just copied over?

Danielauler, 2020-09-22:
I found this solution that have been work for me (https://arcanesciencelab.wordpress.com/2020/04/11/disable-chromium-update-dialog-on-raspbian-buster/)

In the file system under /etc/chromium-browser/customizations, open a new file named 01-disable-update-check:

```
cd /etc/chromium-browser/customizations
sudo vi 01-disable-update-check
```
Then type in the following single line of text:
```
CHROMIUM_FLAGS="${CHROMIUM_FLAGS} --check-for-update-interval=31536000"
```
This isn't my solution, I'm just posting here for who still not found anything to correct this bug

slokhorst, 2020-12-06:
A few days ago, [a new Raspberry Pi OS was released](https://www.raspberrypi.org/blog/new-raspberry-pi-os-release-december-2020/), which contains Chromium 84. That should also include the aforementioned commit https://github.com/chromium/chromium/commit/e79d9d0e06b825d2e62b38db03248c0e6ceec7e4 so the notification shouldn't show ever again.

KTibow, 2020-12-06:
Does that mean we can close this now?

XECDesign, 2020-12-07:
Yup, it should be resolved. Thanks for the nudge.

closed 2020-12-07 by XECDesign