RPi-Distro/repo #166: libgeographic17: Linker error with abi:cxx11 on Buster.

JacekRuzyczka, 2020-02-25:
I'm trying to use GeographicLib 1.47 on Raspbian Buster https://packages.debian.org/buster/libgeographic17 in my C++ app, but I'm always getting a linker error like this:

```
/usr/bin/arm-linux-gnueabihf-ld: main.o: in function `OnDisplayGPSTimerElapsed(P
oco::Util::TimerTask&)': 
/import/valen/autoradio/main.cpp:349: undefined reference to `GeographicLib::DMS
::Encode[abi:cxx11](double, GeographicLib::DMS::component, unsigned int, Geograp
hicLib::DMS::flag, char)' 
/usr/bin/arm-linux-gnueabihf-ld: /import/valen/autoradio/main.cpp:358: undefined
reference to `GeographicLib::DMS::Encode[abi:cxx11](double, GeographicLib::DMS:
:component, unsigned int, GeographicLib::DMS::flag, char)' 
collect2: error: ld returned 1 exit status
```
With an older Raspbian version (Jessie), everything used to work fine. I've found out that the current version of GeographicLib has been compiled with a fairly old version of the GCC (4.9.2), yet the current compiler version of Buster is already 6.5.0:

```
pi@autoradio:/usr/lib $ strings -a libGeographic.so.17.1.1 |grep "GCC: ("        
GCC: (Raspbian 4.9.2-10) 4.9.2 
pi@autoradio:/usr/lib $ g++ --version 
g++ (Raspbian 6.5.0-1+rpi1+b1) 6.5.0 20181026
```

This apparently leads to linking trouble because the C++ 11 ABI has been introduced into the GCC in version 5.1, and type mangling isn't working properly. So: Will anybody recompile the package in question with a GCC >= 5.1, or will I have to compile and pack it myself? Thank you.

XECDesign, 2020-02-26:
The package does not come from archive.raspberrypi.org. You would need to report the issue here https://bugs.launchpad.net/raspbian/

JacekRuzyczka, 2020-02-26:
OK, see https://bugs.launchpad.net/raspbian/+bug/1864920 for further discussion.

closed 2020-02-28 by XECDesign