RPi-Distro/repo #169: Download statistics on curl command line are wrong

Bylon2, 2020-03-15:
Hello, not sure how to file bug into Raspbian packages inherited from Debian.

In the standard Raspbian Buster the curl package is 7.64.0 (as per Debian Buster **Stable**).

This package exhibits a bug when in 32 bits (it is the case here) when a download takes longer than 35m and 47s. It is a signed 32 bit overflow because times are in millseconds the maximum that can be expressed is 2147,483647 seconds, which is 35 minutes 47 seconds and some milliseconds.

I filed the bug at curl here: [https://github.com/curl/curl/issues/5079](https://github.com/curl/curl/issues/5079)

This is a known bug and has been fixed around 7.66

I am currently trying a patch, taken from the commit that fixed it on 7.66, this patch is taken from the commit on Github and seems to apply on 7.64 except the second non important snippet.

**Questions:**
- Is it the right place to report bug on "stable" packages?
- Or should I rather go with 7.68 in "unstable" (not sure how to do that "cleanly" though!)

Here is the patch I am testing. I have build the package successfully including that patch, and I'll let you know if all works well:
```

From b1616dad8f088d873d88f88b4d884335a4ca285f Mon Sep 17 00:00:00 2001
From: Daniel Stenberg <daniel@haxx.se>
Date: Wed, 31 Jul 2019 15:30:31 +0200
Subject: [PATCH] timediff: make it 64 bit (if possible) even with 32 bit
 time_t

... to make it hold microseconds too.

Fixes #4165
Closes #4168
---
 lib/asyn-thread.c |  7 ++++---
 lib/connect.c     | 10 +++++-----
 lib/ftp.c         |  4 ++--
 lib/hostip.c      |  2 +-
 lib/hostip.h      |  3 ++-
 lib/http2.c       |  2 +-
 lib/multi.c       |  4 ++--
 lib/multiif.h     |  2 +-
 lib/pingpong.c    |  4 ++--
 lib/progress.c    | 11 ++++++-----
 lib/select.h      |  6 +++---
 lib/timeval.c     | 24 ++++++++----------------
 lib/timeval.h     | 10 +++++-----
 lib/url.c         |  3 ++-
 lib/urldata.h     | 18 +++++++++---------
 15 files changed, 53 insertions(+), 57 deletions(-)

diff --git a/lib/asyn-thread.c b/lib/asyn-thread.c
index dcc2e8a9d35..222e78d98aa 100755
--- a/lib/asyn-thread.c
+++ b/lib/asyn-thread.c
@@ -603,8 +603,9 @@ CURLcode Curl_resolver_is_resolved(struct connectdata *conn,
   }
   else {
     /* poll for name lookup done with exponential backoff up to 250ms */
-    timediff_t elapsed = Curl_timediff(Curl_now(),
-                                       data->progress.t_startsingle);
+    /* should be fine even if this converts to 32 bit */
+    time_t elapsed = (time_t)Curl_timediff(Curl_now(),
+                                           data->progress.t_startsingle);
     if(elapsed < 0)
       elapsed = 0;

diff --git a/lib/connect.c b/lib/connect.c
index 59084f251dd..be9fcba66c8 100644
--- a/lib/connect.c
+++ b/lib/connect.c
@@ -795,8 +795,8 @@ CURLcode Curl_is_connected(struct connectdata *conn,
     if(rc == 0) { /* no connection yet */
       error = 0;
[... 317 lines cut ...]
   int width; /* screen width at download start */
   int flags; /* see progress.h */
 
-  time_t timespent;
+  timediff_t timespent;
 
   curl_off_t dlspeed;
   curl_off_t ulspeed;
 
-  time_t t_nslookup;
-  time_t t_connect;
-  time_t t_appconnect;
-  time_t t_pretransfer;
-  time_t t_starttransfer;
-  time_t t_redirect;
+  timediff_t t_nslookup;
+  timediff_t t_connect;
+  timediff_t t_appconnect;
+  timediff_t t_pretransfer;
+  timediff_t t_starttransfer;
+  timediff_t t_redirect;
 
   struct curltime start;
   struct curltime t_startsingle;
```

Bylon2, 2020-03-16:
**Follow up:**
The patch above fixes the issue of curl 7.64 that is in the "stable" repo. See linked bug report at curl github.

Now awaiting from Raspbian Gurus to tell us if the right route is either to fix curl 7.64 or use curl 7.68 from "unstable".

XECDesign, 2022-01-19:
Looks like bullseye is on 7.74 so hopefully this one has resolved itself.

closed 2022-01-19 by XECDesign