RPi-Distro/repo #170: Clicking in screen corners doesn't activate menus

Timmmm, 2020-04-23:
I have no idea where to file this bug, sorry. I guess it is in the "PIXEL" desktop but I can't find the code for that anywhere.

Anyway it makes a classic GUI mistake that tons of other Linux desktops make too (only Mac and Windows seem to get this right): If you click in the top-left corner of the screen it doesn't activate the "start" menu (and if you click on the top-right it doesn't activate the button there either).

The hit zones for these buttons should extend all the way to the corner of the screen. Pretty annoying that they don't.

XECDesign, 2020-04-24:
@spl237, I'm sure you're already familiar with it, but https://en.wikipedia.org/wiki/Fitts%27s_law#Implications_for_UI_design

IIRC, last time we tried this, the issue was that point-rpi would break, especially when the panel size and location was adjusted.

spl237, 2020-04-24:
Actually, the main issue is that the menu button doesn't look good unless it is spaced a few pixels from the edge of the screen! I can have a look and see if it is possible to include the space in the button rather than needing a separate spacer; that's probably do-able. We don't use point-rpi any more, so that shouldn't be a problem.

Doing the same for the button in the far right will be non-trivial, because this can be one of several buttons, which would mean having to make all taskbar buttons with integral space. Which I suspect, from having been fiddling with a lot of this stuff last week, will break a lot of things...

XECDesign, 2020-04-24:
Based on my own subjective experience, the right corner is not particularly useful anyway. 

Depending on the OS, that's usually a notification tray, 'show desktop' button or the calendar, none of which are used as frequently as the menu. It's also why I tend to move the panel to the bottom - to make use of the top right corner for closing windows.

But that's all probably more habit than anything else.

spl237, 2020-04-24:
I can probably pad the menu button, although all the sizing code for lxpanel icons is a complete mess, and every time I touch something in it, something else breaks...

I'll put it on the to-do list.

Timmmm, 2020-04-24:
> Actually, the main issue is that the menu button doesn't look good unless it is spaced a few pixels from the edge of the screen!

It isn't necessary to move the button (though I think it should be moved left a little; it looks like too much padding if you ask me). Apple's "Apple" button has padding to the left but it still activates when you click on that padding.

Btw where is the code for this?

spl237, 2020-04-30:
I've modified the menu plugin to incorporate internal padding and thus remove the spacer to its left; this will be in the next Raspbian release,

closed 2020-04-30 by spl237

Timmmm, 2020-05-02:
Nice work, thanks!