RPi-Distro/repo #173: RPi4 on VGA CRT: Picture is shifted off-screen to the left about 1/8 of the screen width using an HDMI to VGA converter on a CRT monitor.

qbancoffee, 2020-05-25:
Description:
After the udev kernel device manager is started, all Raspbian based Os's  shift the picture left about 1/8 of the screen width off-screen. Before that, the rainbow splash screens displays correctly
and the four raspberry's display correctly.
Ubuntu 20.04 64 bit for the Raspberry pi 4 boots and runs correctly, no display issues.

Configuration:
- Raspberry pi 4/4GB Ram
- 4 GB sdcard
- Raspian Lite released 2020-02-13 (burned with raspberry pi imager)
- 1024x768 75Hz Homebrew CRT VGA monitor (Works correctly with Raspian on a Raspberry pi3, 
   x86 linux, windows XP,7 and 10, OS9, OSX and several other os's )
- HDMI to VGA adapater (tested with several different brands)
- Modified config.txt and added the following

```bash
hdmi_group=2
hdmi_mode=18
```

I have tested with all of the available Raspian images from the "Raspberry Pi Imager"
and downloaded and installed RetroPie with the same results. All Raspian based OS's
behave the same way but Ubuntu behaves correctly.

I think this is a moderate bug as many people including myself would like
to do retro gaming on CRT monitors.

Videos demonstrating bug and Ubuntu working correctly:

- [First boot with Raspian Lite](https://youtu.be/X0TvxBZdHI8)
- [Running raspi-config](https://youtu.be/B10Hfp5bqt0)
- [Image shifts when udev kernel device manager is started](https://youtu.be/8rqlzGyEJkE)
- [Ubuntu 20.04 booting correctly on raspberry pi4](https://youtu.be/PDu9LwotDi8)

XECDesign, 2020-05-25:
That should probably be reported here - https://github.com/raspberrypi/firmware/issues

qbancoffee, 2020-05-25:
O.K. just did.

closed 2020-05-25 by XECDesign