RPi-Distro/repo #174: No response to wifi connection

15230745073, 2020-05-26:
Is this the right place for my bug report?
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at github.com/raspberrypi/firmware and Raspberry Pi userland applications at github.com/raspberrypi/userland. If you have problems with the Raspbian distribution packages, report them in the github.com/RPi-Distro/repo. If you simply have a question, then the Raspberry Pi forums are the best place to ask it.

Describe the bug
My 3b + machine burns the latest raspberry debian system. After entering the desktop system, I can search for wifi, but after entering the password, nothing happens. It still shows a red sign, and there is no blue wifi symbol to search. , My other devices use wifi normally

To reproduce
List the steps required to reproduce the issue.

Expected behaviour
Add a clear and concise description of what you expected to happen.

Actual behaviour
Add a clear and concise description of what actually happened.

System
Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

Pi3B+

Which OS and version (cat /etc/rpi-issue)?
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage4
Which firmware version (vcgencmd version)?、
Feb 12 2020 12:39:27
Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start_x)
Which kernel version (uname -a)?
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
Logs
If applicable, add the relevant output from dmesg or similar.
[ 0.000000] Booting Linux on physical CPU 0x0
[ 0.000000] Linux version 4.19.97-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1294 SMP Thu Jan 30 13:15:58 GMT 2020
[ 0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[ 0.000000] CPU: div instructions available: patching division code
[ 0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[ 0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[ 0.000000] Memory policy: Data cache writealloc
[ 0.000000] cma: Reserved 8 MiB at 0x37800000
[ 0.000000] On node 0 totalpages: 229376
[ 0.000000] Normal zone: 2016 pages used for memmap
[ 0.000000] Normal zone: 0 pages reserved
[ 0.000000] Normal zone: 229376 pages, LIFO batch:63
[ 0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[ 0.000000] percpu: Embedded 16 pages/cpu s36864 r8192 d20480 u65536
[ 0.000000] pcpu-alloc: s36864 r8192 d20480 u65536 alloc=16*4096
[ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 227360
[ 0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:04:F8:9E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000 console=ttyS0,115200 console=tty1 root=PARTUUID=ea7d04d6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[ 0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[ 0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[ 0.000000] Memory: 886300K/917504K available (8192K kernel code, 653K rwdata, 2220K rodata, 1024K init, 822K bss, 23012K reserved, 8192K cma-reserved)
[ 0.000000] Virtual kernel memory layout:
vector : 0xffff0000 - 0xffff1000 ( 4 kB)
fixmap : 0xffc00000 - 0xfff00000 (3072 kB)
vmalloc : 0xb8800000 - 0xff800000 (1136 MB)
lowmem : 0x80000000 - 0xb8000000 ( 896 MB)
modules : 0x7f000000 - 0x80000000 ( 16 MB)
.text : 0x(ptrval) - 0x(ptrval) (9184 kB)
.init : 0x(ptrval) - 0x(ptrval) (1024 kB)
.data : 0x(ptrval) - 0x(ptrval) ( 654 kB)
.bss : 0x(ptrval) - 0x(ptrval) ( 823 kB)
[ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[ 0.000000] ftrace: allocating 26860 entries in 79 pages
[ 0.000000] rcu: Hierarchical RCU implementation.
[ 0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[ 0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[ 0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[ 0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[ 0.000020] Switching to timer-based delay loop, resolution 52ns
[... 273 lines cut ...]
[ 324.807006] Under-voltage detected! (0x00050005)
[ 331.047000] rpi_firmware_get_throttled: 7 callbacks suppressed
[ 331.047007] Voltage normalised (0x00000000)
[ 360.167016] Under-voltage detected! (0x00050005)
[ 366.406965] Voltage normalised (0x00000000)
[ 499.526970] Under-voltage detected! (0x00050005)
[ 505.766958] Voltage normalised (0x00000000)
[ 699.207069] rpi_firmware_get_throttled: 1 callbacks suppressed
[ 699.207077] Under-voltage detected! (0x00050005)
[ 705.447022] rpi_firmware_get_throttled: 1 callbacks suppressed
[ 705.447031] Voltage normalised (0x00000000)
[ 790.726981] Under-voltage detected! (0x00050005)
[ 801.126998] Voltage normalised (0x00000000)
[ 923.846973] Under-voltage detected! (0x00050005)
[ 928.006967] Voltage normalised (0x00000000)
[ 1019.526982] Under-voltage detected! (0x00050005)
[ 1023.686969] Voltage normalised (0x00000000)
[ 1152.646979] Under-voltage detected! (0x00050005)
[ 1158.887050] Voltage normalised (0x00000000)
[ 1406.406982] Under-voltage detected! (0x00050005)
[ 1410.566986] Voltage normalised (0x00000000)
[ 1543.686987] Under-voltage detected! (0x00050005)
[ 1549.926985] Voltage normalised (0x00000000)
[ 1699.686992] Under-voltage detected! (0x00050005)
[ 1705.927005] Voltage normalised (0x00000000)

pheiduck, 2020-05-30:
Had a issue like this on Arch Linux ARM; on Pi 2 b connection failed but all was correct. The netctl@ service failed to start at the end. Resolved this by adding `ForceConnect=yes`
IP was assigned and it was reacheable over WIFI

XECDesign, 2022-01-19:
Is this still an issue in Bullseye (and with an adequate power supply)?

closed 2022-02-02 by no-response[bot]