RPi-Distro/repo #175: The certificate of `archive.raspberrypi.org` has expired

PeterLacknase, 2020-05-30:
`sudo apt update`

woes

"https://archive.raspberrypi.org/debian buster Release Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification."

Seems like the *.raspberrypi.org wildcard certificate is still valid but it has to use a different certification path.

[code]Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            13:ea:28:70:5b:f4:ec:ed:0c:36:63:09:80:61:43:36
        Signature Algorithm: sha384WithRSAEncryption
        Issuer: C = SE, O = AddTrust AB, OU = AddTrust External TTP Network, CN = AddTrust External CA Root
        Validity
            Not Before: May 30 10:48:38 2000 GMT
            Not After : May 30 10:48:38 2020 GMT
        Subject: C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority[/code]

XECDesign, 2020-05-30:
Thanks for opening the issue.

For reference: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=275648

XECDesign, 2020-05-30:
Should be sorted.

PeterLacknase, 2020-05-30:
It is. Thanks a lot.

closed 2020-05-30 by XECDesign

edreinoso, 2023-06-09:
Sorry, but I'm persisting with this problem, since it is driving me a bit crazy

I also get a similar message when I'm trying to do a simple: sudo apt upgrade

Certificate verification failed: The certificate is NOT trusted

The difference for me is that my certicate expires in 2028, so there should not be any issues as of why this problem should be arising in the first place.

Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
        Signature Algorithm: sha384WithRSAEncryption
        Issuer: C = GB, ST = Greater Manchester, L = Salford, O = Comodo CA Limited, CN = AAA Certificate Services
        Validity
            Not Before: Mar 12 00:00:00 2019 GMT
            Not After : Dec 31 23:59:59 2028 GMT
        Subject: C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority

XECDesign, 2023-06-13:
Has your system time been updated? You can check with the output of `date`. It takes a while after booting for systemd-timesyncd to do its thing.

edreinoso, 2023-06-13:
Thank you for your reply.

The system time has been updated. You can see the output below:

pi@raspberrypi:~ $ date
Tue 13 Jun 2023 01:53:57 PM CEST
pi@raspberrypi:~ $ timedatectl
               Local time: Tue 2023-06-13 13:55:40 CEST
           Universal time: Tue 2023-06-13 11:55:40 UTC
                 RTC time: n/a
                Time zone: Europe/Amsterdam (CEST, +0200)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no

I have this thread no Raspberry Pi community support for better reference.

https://forums.raspberrypi.com/viewtopic.php?p=2113419#p2113419